robotology-legacy/codyco-modules #96: Investigate random bug of wholeBodyDynamicsTree

traversaro, 2015-02-27:
```
iDynTree WARNING: base_residual_f.force.Norm() is 0.0021643 instead of zero.
iDynTree WARNING: base_residual_f.torque.Norm() is 0.00101268 instead of zero.
```

traversaro, 2015-02-28:
More info:
~~~
528		39.728312				wholeBodyDynamicsTree: double support calibration for allrequested
529		39.728390				wholeBodyDynamicsThread::calibrateOffsetOnDoubleSupport called with code all
530		39.728425				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 0 is -nan -nan -nan -nan -nan -nan
531		39.728456				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 1 is -nan -nan -nan -nan -nan -nan
532		39.728486				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 2 is -nan -nan -nan -nan -nan -nan
533		39.728515				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 3 is 49.8407 5.2716 15.6721 -0.789865 0.13584 -0.280371
534		39.728545				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 4 is -nan -nan -nan -nan -nan -nan
535		39.728573				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 5 is -nan -nan -nan -nan -nan -nan
536		39.728601				wholeBodyDynamicsThread::calibrateOffset all called successfully, starting calibration.
551		43.103568				wholeBodyDynamicsTree: double support calibration for allrequested
552		43.103646				wholeBodyDynamicsThread::calibrateOffsetOnDoubleSupport called with code all
553		43.103680				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 0 is 0 0 0 0 0 0
554		43.103708				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 1 is 0 0 0 0 0 0
555		43.103734				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 2 is 0 0 0 0 0 0
556		43.103760				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 3 is 0 0 0 0 0 0
557		43.103786				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 4 is 0 0 0 0 0 0
558		43.103812				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 5 is 0 0 0 0 0 0
707		67.776289				yarp: Removing output from /wholeBodyDynamicsTree/contacts:o to /iCubGui/forces
709		68.048372				yarp: Removing output from /wholeBodyDynamicsTree/base:o to /iCubGui/base:i
730		69.015886				iDynTree WARNING: base_residual_f.force.Norm() is 0.226746 instead of zero.
731		69.015927				iDynTree WARNING: base_residual_f.torque.Norm() is 0.0605821 instead of zero.
~~~

It is quite dangerous, because it corrupts all torque estimates.

traversaro, 2015-03-02:
```
5292		1931.365924				||| clearing context
5293		1931.371788				||| adding context [wholeBodyDynamicsTree]
5294		1931.371849				||| configuring
5295		1931.371908				||| no policy found
5296		1931.371938				||| default config file specified as wholeBodyDynamicsTree.ini
5297		1931.371969				||| checking [/tmp/wholeBodyDynamicsTree.ini] (pwd)
5298		1931.371997				||| checking [/home/icub/.config/yarp/robots/iCubParis01] (robot YARP_CONFIG_HOME)
5299		1931.372027				||| checking [/home/icub/.local/share/yarp/robots/iCubParis01] (robot YARP_DATA_HOME)
5300		1931.372056				||| found /home/icub/.local/share/yarp/robots/iCubParis01
5301		1931.372091				||| checking [/etc/yarp/robots/iCubParis01] (robot YARP_CONFIG_DIRS)
5302		1931.372121				||| checking [/home/icub/software/src/yarp/_build/share/yarp/robots/iCubParis01] (robot YARP_DATA_DIRS)
5303		1931.372152				||| checking [/home/icub/software/src/icub-main/_build/share/iCub/robots/iCubParis01] (robot YARP_DATA_DIRS)
5304		1931.372184				||| found /home/icub/software/src/icub-main/_build/share/iCub/robots/iCubParis01
5305		1931.372249				||| checking [/home/icub/software/src/codyco-superbuild/_build/install/share/codyco/robots/iCubParis01] (robot YARP_DATA_DIRS)
5306		1931.372281				||| found /home/icub/software/src/codyco-superbuild/_build/install/share/codyco/robots/iCubParis01
5307		1931.372312				||| checking [/home/icub/software/src/yarp/_build/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
5308		1931.372352				||| checking [/home/icub/software/src/icub-main/_build/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
5309		1931.372384				||| checking [/home/icub/software/src/codyco-superbuild/_build/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
5310		1931.372413				||| checking [/home/icub/.local/share/yarp/robots/iCubParis01/wholeBodyDynamicsTree.ini] (robot)
5311		1931.372443				||| found /home/icub/.local/share/yarp/robots/iCubParis01/wholeBodyDynamicsTree.ini
5312		1931.372492				assume_fixed option found, using l_sole as fixed link as a kinematic root instead of the imu.
5313		1931.372529				assume_fixed_base_calibration option found, using l_sole as fixed link as a kinematic root instead of the imu for calibration.
5314		1931.372560				yarp: Port /wholeBodyDynamicsTree/rpc:i active at tcp://10.0.0.13:10175
5315		1931.372590				||| finding file [wbi_conf_file]
5316		1931.372617				||| checking [/tmp/yarpWholeBodyInterface.ini] (pwd)
5317		1931.372645				||| checking [/home/icub/.local/share/yarp/robots/iCubParis01/yarpWholeBodyInterface.ini] (robot)
5318		1931.372674				||| found /home/icub/.local/share/yarp/robots/iCubParis01/yarpWholeBodyInterface.ini
5319		1931.372703				 loadIdListFromConfig error: requested list ROBOT_DYNAMIC_MODEL_JOINTS not found
5323		1931.386079				yarp: Port /wholeBodyDynamicsTree/skin_contacts:i active at tcp://10.0.0.13:10176
5333		1931.486926				yarp: Port /wholeBodyDynamicsTree/torso/rpc:o active at tcp://10.0.0.13:10177
5482		1931.547580				yarp: Port /wholeBodyDynamicsTree/torso/command:o active at tcp://10.0.0.13:10178
5577		1932.496406				yarp: Port /wholeBodyDynamicsTree/torso/state:i active at tcp://10.0.0.13:10179
5694		1932.627378				yarp: Port /wholeBodyDynamicsTree/torso/stateExt:i active at tcp://10.0.0.13:10180
5699		1932.768583				yarp: Sending output from /wholeBodyDynamicsTree/torso/rpc:o to /icub/torso/rpc:i using tcp
5715		1932.890215				yarp: Sending output from /wholeBodyDynamicsTree/torso/command:o to /icub/torso/command:i using tcp
5724		1933.101843				yarp: Receiving input from /icub/torso/state:o to /wholeBodyDynamicsTree/torso/state:i using udp
5733		1933.152326				yarp: Receiving input from /icub/torso/stateExt:o to /wholeBodyDynamicsTree/torso/stateExt:i using udp
5734		1933.162262				RemoteControlBoard is ENABLING the writeStrict option for all commands
5735		1933.162332		INFO		created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
5739		1933.242947				yarp: Port /wholeBodyDynamicsTree/left_arm/rpc:o active at tcp://10.0.0.13:10185
5743		1933.283331				yarp: Port /wholeBodyDynamicsTree/left_arm/command:o active at tcp://10.0.0.13:10186
5745		1933.313577				yarp: Port /wholeBodyDynamicsTree/left_arm/state:i active at tcp://10.0.0.13:10187
5747		1933.384231				yarp: Port /wholeBodyDynamicsTree/left_arm/stateExt:i active at tcp://10.0.0.13:10188
5756		1933.404636				yarp: Sending output from /wholeBodyDynamicsTree/left_arm/rpc:o to /icub/left_arm/rpc:i using tcp
5764		1933.414649				yarp: Sending output from /wholeBodyDynamicsTree/left_arm/command:o to /icub/left_arm/command:i using tcp
5771		1933.434650				yarp: Receiving input from /icub/left_arm/state:o to /wholeBodyDynamicsTree/left_arm/state:i using udp
5773		1933.434676				yarp: Receiving input from /icub/left_arm/stateExt:o to /wholeBodyDynamicsTree/left_arm/stateExt:i using udp
5774		1933.434690				RemoteControlBoard is ENABLING the writeStrict option for all commands
5775		1933.434704		INFO		created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
5776		1933.464797				yarp: Port /wholeBodyDynamicsTree/right_arm/rpc:o active at tcp://10.0.0.13:10189
5777		1933.495038				yarp: Port /wholeBodyDynamicsTree/right_arm/command:o active at tcp://10.0.0.13:10190
5800		1933.535315				yarp: Port /wholeBodyDynamicsTree/right_arm/state:i active at tcp://10.0.0.13:10191
[... cut ...]
43749		6999.142583				iDynTree WARNING: base_residual_f.force.Norm() is 0.380378 instead of zero.
43750		6999.142628				iDynTree WARNING: base_residual_f.torque.Norm() is 0.935381 instead of zero.
43761		7001.589404				iDynTree WARNING: base_residual_f.force.Norm() is 0.999908 instead of zero.
43762		7001.589459				iDynTree WARNING: base_residual_f.torque.Norm() is 0.786293 instead of zero.
43791		7005.112922				iDynTree WARNING: base_residual_f.force.Norm() is 0.626621 instead of zero.
43792		7005.112948				iDynTree WARNING: base_residual_f.torque.Norm() is 0.972957 instead of zero.
```

traversaro, 2015-03-02:
```
96647		14573.726812				wholeBodyDynamicsThread::calibrateOffsetOnDoubleSupport called with code all
96648		14573.726847				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 0 is 88.2863 -50.366 -73.8974 -1.75059 -2.14232 -0.318741
96649		14573.726878				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 1 is 29.4178 -31.6457 16.1286 -0.414776 -1.4949 0.300277
96650		14573.726906				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 2 is -66.4405 90.6395 -34.1493 -2.65656 3.55524 -0.459693
96651		14573.726934				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 3 is 50.3886 7.86793 8.30326 -0.710595 0.488621 -0.0300305
96652		14573.726963				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 4 is -16.4742 -21.7289 -12.2146 2.86362 -2.14509 -1.04518
96653		14573.726991				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 5 is -89.7327 -32.7603 12.5575 -1.16617 2.25775 -1.38008
96654		14573.727019				wholeBodyDynamicsThread::calibrateOffset all called successfully, starting calibration.
96691		14579.769476				wholeBodyDynamicsThread: new calibration for FT l_arm_ft_sensor is 88.1561 -50.8234 -74.0591 -1.75623 -2.14354 -0.316833
96692		14579.769552				wholeBodyDynamicsThread: new calibration for FT r_arm_ft_sensor is 29.522 -31.3741 15.9098 -0.410898 -1.50436 0.286432
96693		14579.769587				wholeBodyDynamicsThread: new calibration for FT l_leg_ft_sensor is -69.7649 88.9746 -32.1717 -2.64402 2.04259 -0.171214
96694		14579.769616				wholeBodyDynamicsThread: new calibration for FT l_foot_ft_sensor is 48.4618 6.63787 10.8762 -0.592095 0.0226885 0.218741
96695		14579.769644				wholeBodyDynamicsThread: new calibration for FT r_leg_ft_sensor is -11.9311 -23.8294 -14.3976 2.95628 -0.651918 -0.664938
96696		14579.769672				wholeBodyDynamicsThread: new calibration for FT r_foot_ft_sensor is -87.5081 -34.9048 10.3362 -1.08757 2.61911 -1.06188
96697		14579.769700				wholeBodyDynamicsThread::waitCalibrationDone() returning: calibration finished with success
97975		14751.292269				wholeBodyDynamicsTree: double support calibration for allrequested
97976		14751.292356				wholeBodyDynamicsThread::calibrateOffsetOnDoubleSupport called with code all
97977		14751.292389				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 0 is 88.1561 -50.8234 -74.0591 -1.75623 -2.14354 -0.316833
97978		14751.292420				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 1 is 29.522 -31.3741 15.9098 -0.410898 -1.50436 0.286432
97979		14751.292452				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 2 is -69.7649 88.9746 -32.1717 -2.64402 2.04259 -0.171214
97980		14751.292481				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 3 is 48.4618 6.63787 10.8762 -0.592095 0.0226885 0.218741
97981		14751.292525				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 4 is -11.9311 -23.8294 -14.3976 2.95628 -0.651918 -0.664938
97982		14751.292554				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 5 is -87.5081 -34.9048 10.3362 -1.08757 2.61911 -1.06188
97983		14751.292583				wholeBodyDynamicsThread::calibrateOffset all called successfully, starting calibration.
98020		14757.333646				wholeBodyDynamicsThread: new calibration for FT l_arm_ft_sensor is 87.9276 -50.5468 -74.1263 -1.7554 -2.15693 -0.312316
98021		14757.333725				wholeBodyDynamicsThread: new calibration for FT r_arm_ft_sensor is 29.5668 -31.6658 16.1857 -0.402014 -1.51447 0.273345
98022		14757.333760				wholeBodyDynamicsThread: new calibration for FT l_leg_ft_sensor is -71.2937 91.0444 -32.1873 -2.76949 2.0004 -0.564645
98023		14757.333790				wholeBodyDynamicsThread: new calibration for FT l_foot_ft_sensor is 46.4645 8.03456 10.9253 -0.600317 0.804351 -0.110612
98024		14757.333827				wholeBodyDynamicsThread: new calibration for FT r_leg_ft_sensor is -9.71795 -26.0739 -14.2932 3.00568 -0.565145 -0.039529
98025		14757.333856				wholeBodyDynamicsThread: new calibration for FT r_foot_ft_sensor is -86.3304 -37.1381 10.6872 -1.08752 2.15941 -0.386666
98026		14757.333887				wholeBodyDynamicsThread::waitCalibrationDone() returning: calibration finished with success
98986		14868.487111				wholeBodyDynamicsTree: double support calibration for allrequested
98987		14868.487164				wholeBodyDynamicsThread::calibrateOffsetOnDoubleSupport called with code all
98988		14868.487190				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 0 is 87.9276 -50.5468 -74.1263 -1.7554 -2.15693 -0.312316
98989		14868.487212				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 1 is 29.5668 -31.6658 16.1857 -0.402014 -1.51447 0.273345
98990		14868.487232				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 2 is -71.2937 91.0444 -32.1873 -2.76949 2.0004 -0.564645
98991		14868.487252				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 3 is 46.4645 8.03456 10.9253 -0.600317 0.804351 -0.110612
98992		14868.487271				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 4 is -9.71795 -26.0739 -14.2932 3.00568 -0.565145 -0.039529
98993		14868.487291				wholeBodyDynamicsThread::calibrateOffset: current calibration for FT 5 is -86.3304 -37.1381 10.6872 -1.08752 2.15941 -0.386666
98994		14868.487309				wholeBodyDynamicsThread::calibrateOffset all called successfully, starting calibration.
[... cut ...]
135133		20757.925759				yarpWholeBodyStatesLocal::close() : deleting sensor interface
135134		20757.925773				yarpWholeBodyStatesLocal::close() : deleting estimator thread
135135		20757.925787				wholeBodyDynamicsTree: closing RPC port interface
135136		20757.935712		INFO		:
135137		20757.935748				Expected period 10 ms.
135138		20757.935763				Real period: 10.1+/-0.2 ms.
135139		20757.935776				Real duration of 'run' method: 0.4+/-0.5 ms.
135140		20757.935790				Next time you could set a lower period to improve the wholeBodyDynamics performance.
```

DanielePucci, 2015-03-02:
+1 very well

traversaro, 2015-03-23:
More examples: https://gist.github.com/traversaro/b745fc4e8a82db581f01 .

traversaro, 2015-06-15:
Just found a way of replicating this: just exert two forces with different application points on a single link. Working on a fix.

traversaro, 2015-06-15:
Bug fixed in https://github.com/robotology-playground/idyntree/commit/b6985e807fc074b4763b7b60eb0a0a37ec4a50b8 . 
A slight mismatch remain in the base residual force, but I guess is something related to some kind of one-step offset between the value readed from the robot. However it is a relatively minor problem, addresed in issue https://github.com/robotology/codyco-modules/issues/122 .

closed 2015-06-15 by traversaro