robotology-legacy/codyco-modules #106: [torqueBalancing] Add an automatic calibration of wholeBodyDynamicsTree befor starting the module

DanielePucci, 2015-03-16:
For the time being, we run the calibration of WBDT before the start of the module (manually and esternally), and this calibration is designed ad-hoc for the robot to stand on both feet. Can we add an automatic calibration of WBDT before the start of the torqueBalancing? @francesco-romano @traversaro

DanielePucci, 2015-04-02:
Any ideas on this?

DanielePucci, 2015-10-11:
ping

traversaro, 2015-10-11:
What to do you mean "add an automatic calibration of WBDT before the start of the torqueBalancing" ? The only thing that we can do is to have the torqueBalancing (even if strictly speaking it would be more appropriate to have some kind of "supervisor" to do this) to call via rpc the WBDT and launch a calibration, but I don't see how you can move away from the "calibration is designed ad-hoc for the robot to stand on both feet".

DanielePucci, 2015-10-11:
Yes, this is what I meant: rpc command. It could be nice to have an option in torque balancing like

~~~
---calibAtStartTwoFeet
~~~

or
~~~
---calibAtStartLeftFoot
~~~

which runs an rpc command before starting. What do you think about it? @traversaro @francesco-romano

francesco-romano, 2015-10-11:
As you know I don’t like this idea

DanielePucci, 2015-10-11:
Which idea? That of the external coordinator calibration or that of the rpc command? Assuming that you do not like the idea of the option, any alternative?

traversaro, 2019-03-15:
The torqueBalancing module has not been used since a long time, I think it just make sense to close all the issues related to it.

closed 2019-03-15 by traversaro