robotology-legacy/codyco-modules #112: [wholeBodyDynamicsTree] 3 threads saturate CPU usage when Yarp is compiled with OpenBLAS

traversaro, 2015-03-24:
On the `iCubGenova01` setup, we notice that wholeBodyDynamicsTree was unusually CPU-intensive, after a quick debug we realized that 3 threads where actually saturating a core each:

```
top - 10:09:17 up 19 days, 20:33,  5 users,  load average: 0.86, 0.21, 0.11
Threads: 428 total,   5 running, 354 sleeping,   0 stopped,   0 zombie
%Cpu(s):  3.2 us, 17.6 sy,  0.0 ni, 79.2 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem:  16408140 total,  1280408 used, 15127732 free,        0 buffers
KiB Swap:        0 total,        0 used,        0 free,   556884 cached

  PID USER      PR  NI  VIRT  RES  SHR S  %CPU %MEM    TIME+  COMMAND                                                                                                                                           
13364 icub      20   0 1065m 159m  10m R  99.9  1.0   0:17.54 wholeBodyDynami                                                                                                                                   
13366 icub      20   0 1065m 159m  10m R  99.9  1.0   0:17.55 wholeBodyDynami                                                                                                                                   
13365 icub      20   0 1065m 159m  10m R  99.6  1.0   0:17.54 wholeBodyDynami                                                                                                                                   
13461 icub      20   0 1065m 159m  10m R  13.9  1.0   0:02.59 wholeBodyDynami                                                                                                                                   
13391 icub      20   0 1065m 159m  10m S   3.6  1.0   0:00.76 wholeBodyDynami                                                                                                                                   
13389 icub      20   0 1065m 159m  10m S   2.6  1.0   0:00.49 wholeBodyDynami                                                                                                                                   
13385 icub      20   0 1065m 159m  10m S   1.7  1.0   0:00.36 wholeBodyDynami                                                                                                                                   
13384 icub      20   0 1065m 159m  10m S   1.3  1.0   0:00.37 wholeBodyDynami                                                                                                                                   
13403 icub      20   0 1065m 159m  10m S   0.7  1.0   0:00.13 wholeBodyDynami                                                                                                                                   
13409 icub      20   0 1065m 159m  10m S   0.7  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13415 icub      20   0 1065m 159m  10m S   0.7  1.0   0:00.12 wholeBodyDynami                                                                                                                                   
13475 icub      20   0 22044 1836 1072 R   0.7  0.0   0:00.05 top                                                                                                                                               
12360 icub      20   0 22052 1836 1076 S   0.3  0.0   0:14.35 top                                                                                                                                               
13373 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13379 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.07 wholeBodyDynami                                                                                                                                   
13396 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13397 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13401 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.05 wholeBodyDynami                                                                                                                                   
13408 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13420 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13421 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13425 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.05 wholeBodyDynami                                                                                                                                   
13427 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.08 wholeBodyDynami                                                                                                                                   
13432 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13433 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13437 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.05 wholeBodyDynami                                                                                                                                   
13439 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.08 wholeBodyDynami                                                                                                                                   
13445 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13448 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13451 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13454 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.04 wholeBodyDynami                                                                                                                                   
13460 icub      20   0 1065m 159m  10m S   0.3  1.0   0:00.05 wholeBodyDynami                                                                                                                                   
    1 root      20   0 10644  816  688 S   0.0  0.0   0:14.48 init                                                                                                                                              
    2 root      20   0     0    0    0 S   0.0  0.0   0:00.02 kthreadd                                                                                                                                          
    3 root      20   0     0    0    0 S   0.0  0.0   0:02.16 ksoftirqd/0  
```   

Apparently this three threads are created at loading by OpenBlas : 
~~~
(gdb) thread 2   
[Switching to thread 2 (Thread 0x7ffff0726700 (LWP 2123))]
#0  0x00007ffff13f9fb7 in sched_yield () at ../sysdeps/unix/syscall-template.S:82
82	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) backtrace
#0  0x00007ffff13f9fb7 in sched_yield () at ../sysdeps/unix/syscall-template.S:82
#1  0x00007ffff308d9e5 in ?? () from /usr/lib/libopenblas.so.0
[... cut ...]
This is the system in which we experienced this issue:
~~~
icub@icub15:/usr/local/src/robot/codyco-superbuild/build-x86_64 (master)$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.6 (wheezy)
Release:	7.6
Codename:	wheezy
~~~

And this is the installed version of OpenBLAS:
~~~
icub@icub15:/usr/local/src/robot/codyco-superbuild/build-x86_64 (master)$ apt-cache show libopenblas-base
Package: libopenblas-base
Source: openblas
Version: 0.1.1-6+deb7u3
Installed-Size: 19138
~~~

traversaro, 2015-07-22:
cc @francesco-romano @alecive

francesco-romano, 2015-11-14:
OpenBLAS has been removed on the black setup.
I think we can close this issue.

closed 2015-11-14 by francesco-romano

DanielePucci, 2015-11-14:
True. On the other hand, it still happens that sometimes you go there and OpenBlas has been installed, causing an increase of the CPU use, e.g. IROS 2015. We should put here the proceed to uninstall it.