robotology-legacy/codyco-modules #114: [TorqueBalancing] Clean up module

francesco-romano, 2015-03-31:
The module was thought with the CoDyCo Y1 demo in mind.
New requirements now ask for cleaning up the module and transform it into a generic balancing module

Tasks are:

- [x] create and reads external references through RPC port
- [ ] remove the old embedded FSM
- [ ] handle contraints changes

traversaro, 2015-03-31:
* [ ] adding an `events:i` port to raise events  (to be discussed) ?

francesco-romano, 2015-03-31:
`events:o` ? Anyway, +1

traversaro, 2015-03-31:
Just for refence: the place holder functions for activating/deactivating contacts in lua are: https://github.com/robotology/codyco-modules/blob/master/src/modules/graspAndStepDemo/lua/gas_funcs.lua#L28 . Obviously we are free to change them.

traversaro, 2015-03-31:
* [ ] remove references to the "hand" end-effectors (until we have the link-generic constraint support)?

traversaro, 2015-03-31:
ref https://github.com/robotology/codyco-superbuild/issues/61

francesco-romano, 2015-11-14:
All the tasks have been completed, or they are no longer needed. 
Closing this issue.

closed 2015-11-14 by francesco-romano