robotology-legacy/codyco-modules #118: Get rid of codyco-modules/extern/eigen_unsupported

traversaro, 2015-05-11:
In this repository there is a directory that is hosting the `Eigen` LGSM module. 
Given that the ocra modules are not getting LGSM from here, I guess it is safe to remove it? 
@rlober @serena-ivaldi

traversaro, 2015-05-11:
Can we also get rid of the `src/python` directory? It's use is not explained anywhere, and they are messing up doxygen documentation.

rlober, 2015-05-11:
As far as I am concerned **yes** for `eigen_unsupported`. I have created a repo for our old eigen 3.0.5 with lgsm [link to repo](https://github.com/ocra-recipes/eigen_3_0_5). So even if you delete it we have a backup. 

As for `src/python` it doesn't look like it has seen much action recently, but since these are all Serena's I will defer to her. My guess is that she doesn't need them anymore, but it may be good to make a backup.

traversaro, 2016-06-07:
Done in https://github.com/robotology/codyco-modules/pull/185 .

closed 2016-06-07 by traversaro