robotology-legacy/codyco-modules #123: automatic calibration of foot accelerometer / gyro rototranslation

naveenoid, 2015-06-18:
In the module quaternionEKF, the calibration procedure must exist within the module as a C++ component without having the need to use Matlab. quaternionEKF also needs debugging for the singularity issue.

traversaro, 2015-06-23:
My two cents: 
  * I don't know if C++ is the right way to go... for friction estimation we noticed that online inspection of results is fundamental for a day-to-day use of a calibration tool, and with C++ it is quite tricky to implement. Perhaps some (free) scripting language is more indicated? 
 * For storing calibration results, we need a format to store sensors information.. we can proceed following the original idea in https://github.com/robotology-playground/idyntree/issues/30 or using some simpler model, but I think it is important that we agree on some format and shared it, instead of having information spread to N mutually incompatible configuration files.

randaz81, 2015-06-23:
You can look at https://github.com/robotology/icub-tests/tree/master/src/opticalEncoders-consistency for an example of plot in C++ (using gnuplot). Actually a https://github.com/robotology/icub-tests/tree/master/src/motor-stiction is also available, but is not fully finished.

traversaro, 2015-06-23:
Perfect then, we can go with C++ with proper plotting.

naveenoid, 2015-11-28:
closing this for now..will reopen once wholeBodyEstimator requires something like this.

closed 2015-11-28 by naveenoid