robotology-legacy/codyco-modules #128: [wholeBodyDynamicsTree] The parameter "cutoff" in wholeBodyDynamics.ini is ineffective

DanielePucci, 2015-07-29:
The parameter "cutoff" in the configuration file wholeBodyDynamicsTree.ini is ineffective.

traversaro, 2015-07-29:
Actually it is effective, it changes the cutoff frequency of the only low-pass filter that is executed in the wholeBodyDynamicsTree, the one that filters IMU measurements.

DanielePucci, 2015-07-29:
Then, as we discussed, changing the name of the parameter may help users to understand better its meaning. Otherwise, one may be confused by its name, and assume that other measurements arriving to the module, e.g. those from the FT sensors, are also filtered.

traversaro, 2015-07-29:
`cutoff` was documented both in the help message and in the doxygen documentation (not updated due to https://github.com/robotology/codyco-superbuild/issues/80 ). 
By the way, this issue should be solved by https://github.com/robotology/codyco-modules/pull/129 .. can you check it ?

DanielePucci, 2015-07-30:
It seems to me that in the help message, it was not specified that ~~cutoff~~ was for the IMU measurements only, i.e.

~~~
[INFO]	--cutoff           :cutoff frequency (in Hz) of the low pass filters used in wholeBodyDynamicsTree. 
~~~

Anyways, I'll check the fix soon. Thanks a lot Si :+1:

traversaro, 2015-07-30:
The only filters in the module where the ones for the IMU measurements, so the documentation was correct (and consistent with the old wbd). I was wrong in remembering the filter that where implemented in the module.

DanielePucci, 2015-07-30:
I checked the effect of 
~~~
cutoff_ft 
~~~
while performing a trajectory tracking (qd(t) = 10° sin(0.5Hz t) ) through a computed-torque control.

Here are the results.
~~~
No cutoff_ft in wholeBodyDynamicsTree.ini 
~~~
R_leg FT measurements

[image]

Torque tracking

[image]

~~~
cutoff_ft 3.0
~~~

R_leg FT measurements

[image]

Torque tracking

[image]

closed 2015-07-30 by traversaro