robotology-legacy/codyco-modules #130: Compliant foot planting on angled surfaces

naveenoid, 2015-08-07:
Issue to track the tasks needed to investigate compliant foot planting under uncertainty on surface orientation.
[ ] - Cartesian solver for the 2 legs to move from double stable phase to single foot planting
[ ] - Exposing of impedance control interface
[ ] - Setup of gazebo with the angled surface block object
[ ] - Setup of experiment data gathering scripts

traversaro, 2015-08-07:
For the first bullet point, please check : https://github.com/robotology/codyco-modules/issues/107 .

francesco-romano, 2015-08-07:
Second point has a separate issue: https://github.com/robotology-playground/yarp-wholebodyinterface/issues/41

francesco-romano, 2015-08-07:
Interaction mode implemented for a single joint in https://github.com/robotology-playground/yarp-wholebodyinterface/commit/8a24605ef3956591edba90a99687a73abeed0245

traversaro, 2017-08-10:
Research actively successfully published in  http://ieeexplore.ieee.org/abstract/document/7803270/ , closing the issue.

closed 2017-08-10 by traversaro