robotology-legacy/codyco-modules #131: [quaternionEKF] TODO List for preliminary implementation on iCub with palm skin patch

jeljaik, 2015-09-10:
- [ ] Get palm skin patch with gyro and acc installed on right foot and working on the purple robot. i.e. reading accessible from yarp port `/icub/right_leg/inertialMTB`.
- [ ] Fix module in branch quaternionEKF/iCubImplementation to work accordingly.
- [ ] Review calibration.

jeljaik, 2015-09-23:
#### Remove dependencies on XSens classes and substitute with xsens yarp plugin
It was pointed out to me that I could use the *XSens plugin* in YARP when attaching a single `xsens` IMU to my laptop. It is as simple as adding to my `quaternionEKF` application the following module:

```bash
yarpdev --device inertial --subdevice xsensmtx --name /externalxsens/data:o
```
Next, you will find a port called `/externalxsens/data:o` with a similar output to that of `/inertial` when using the real robot. 

*Simplification done in commit:*

jeljaik, 2015-10-14:
## Notes on current status of the palm patch attached to the right foot of the purple robot

The temporary solution that has been found to be able to have a gyroscope somewhere in the foot along with an accelerometer, was to use a skin palm of the iCub and attach it on top of the F/T sensor of the right foot. The way how this works is by attaching this palm to one of the boards of the foot (There are currently two present with their own accelerometers that we are **not** using in this particular firmware mod). The ethernet robot has some differences with respect to the black robot as to how to add new hardware, mainly due to some intermediate element between the new sensor that you connect and the ethernet network. Marco Accame was kind enough to help me handle this issue, by compiling a specific version of the firmware that enables the streaming of the palm sensors (acc + gyro) to the ethernet. This means, however, that if a new firmware is loaded on the boards of the feet (or the whole robot), you must make sure that it's merged in the new firmware, and then load it manually. Let me describe how and where you can do this.

In PC104:
1. `cd` into `/usr/local/robot/icub-firmware-build/testing`
2. `git pull` to fetch the most recent version of our specific feet firmware.
3. From terminal launch `ethLoader` :arrow_right_hook: Discover :arrow_right_hook: Click on 10.0.1.10 (IP for right foot) and 10.0.1.11 :arrow_right_hook: Click on Procs :arrow_right_hook: check the last version is similar to the original one, e.g. current firmware version is 2.10, then ours will be 2**00**.10. :arrow_right_hook: click on Upload App :arrow_right_hook: browse to  `.../icub-firmware-build/testing/em-jorhabib.hex` :arrow_right_hook: Click on Open.

At this point you have loaded a merged version (with what we need) of the most recent firmware only for the two feet. 

Then, in order to enable the gyroscope from this board do the following:

In PC104
1. `cd` into `$ICUB_DIR/share/iCub/robots/iCubGenova02/hardware/skin`
2. Modify the file `right_leg-ems11-inertial_mtb.xml` and in the field `param name = enabledGyroscopes` put the corresponding board to which the palm is attached. Either `r_foot_1` or `r_foot_2`

Now you're good to go and can proceed to start the `robotInterface`

The procedure is currently very cumbersome but this is seen by the firmware team as a temporary solution, on one hand because the feet boards should in the future contain also a gyroscope, on the other, because we still need to have something solid that we are actually using before introducing permanently these changes. An option that I suggested to Mr. Accame was to enable these options on the firmware from configuration file. He will look into this option after making some changes to the code and I will try to remind him, so that he doesn't have to compile a version for us everytime the firmware changes. 

cc. @naveenoid @prashanthr05 @traversaro @iron76

jeljaik, 2015-10-15:
I guess the codyco-modules wiki might be appropriate.

traversaro, 2017-08-10:
Closing for cleanup of old issues.

closed 2017-08-10 by traversaro