robotology-legacy/codyco-modules #132: [torqueBalancing] Make external module for generating references

DanielePucci, 2015-09-21:
For the time being, the references for desired positions for CoM and postural tasks are provided by an external module running in Simulink. This module should be ported in C++ if we do not come up with something smarter from the architecture point of view. It seems to me that the solution in Lua is slowing dying. @traversaro @francesco-romano

traversaro, 2015-09-21:
The Lua part is still a valid option (and the way to go in my opinion) for FSM handling, using rFSM as we do in the rest of complex iCub demos. The thing that I believe was an error was to put too much external logic in the lua script that handled the FSM, and that is not maintainable.

DanielePucci, 2015-09-23:
After a chat with @francesco-romano, we've decided that the simple left-and-right two-feet balancing and one-foot yoga demo may be coordinated by a simple C++ module. The idea is to write a module that streams the desired position of the CoM and the postural task references. The left-and-right two-feet balancing can be achieved by implementing a sine time-function of given amplitude and frequency. The yoga demo, instead, loads from the configuration file a time-series of postures. An idea may be:

~~~
postures = ( timePostureStarts0, q0
                    timePostureStarts1, q1 ) 
~~~
 
To define which data to stream, we may also add an option such as

~~~
type("com","postural")
~~~

DanielePucci, 2015-10-01:
https://github.com/robotology/codyco-modules/commit/2b3b64fbfd48396bf358e123849749cf13baae69

closed 2015-10-01 by DanielePucci