robotology-legacy/codyco-modules #133: [wholeBodyState Module] Planification

jeljaik, 2015-09-24:
## Brief Description

This issue's main purpose is to keep track of the design decisions and specifications for a new module, so far called `wholeBodyState` (are we sure?) belonging to `codyco-modules` and that will exploit the latest improvements on the `wholeBodySensors` interface class, the `BFL` and non-local parametrizations of the orientation. Once all the main tasks have been achieved 

*cc*: @traversaro @iron76 @naveenoid @DanielePucci 

## Design Specifications
- [ ] First off ... what name do we actually wanna give to this module? `wholeBodyState` sounds more like a class belonging to `wholeBodyInterface`
    - [ ] `wholeBodyState`
    - [ ] `wholeBodyStateEstimator`
    - [ ] `wholeBodyStateModule`

DanielePucci, 2015-09-24:
I remember that the name was:

~~~
wholeBodyStateEstimator
~~~

jeljaik, 2015-09-24:
### (T01) 
Do we all agree `wholeBodyStateEstimator` shall be the name?

DanielePucci, 2015-09-24:
Shall me make a pool?

traversaro, 2015-09-24:
I agree on `wholeBodyEstimator`

DanielePucci, 2015-09-24:
:+1: 
~~~
wholeBodyEstimator
~~~

traversaro, 2015-10-10:
I have updated the design doc with some links to existing issues that are a prerequisite to properly implement the modules. 
For the URDF --> Sensors issue in iDynTree ( https://github.com/robotology/idyntree/issues/30 ) @naveenoid is the person in charge, that I guess can deliver an estimate on when the feature will be ready. 
The other important task is https://github.com/robotology/yarp-wholebodyinterface/issues/42 , and I don't know if @jeljaik or @naveenoid want to try to solve it.

iron76, 2015-10-13:
Isn't this related also to the following issue? https://github.com/robotology/idyntree/issues/78. I am asking this since @DanielePucci needs this last issue to be solved (to implement the contact switching demo) and none is currently assigned to it.

jeljaik, 2015-10-13:
@traversaro while narrowing down the actual scope of this module I came to wonder whether the floating base should be an output of this module, or if some other module will exploit `wholeBodyState` to do so. You mentioned something about the `singleLeggedOdometry` class mentioned by nori (https://github.com/robotology/idyntree/issues/78)

traversaro, 2015-10-13:
My original idea was that all estimation algorithms could run sequentially in the `run()` of the thread of the `wholeBodyState` module (similar to what is happening now in [`wholeBodyDynamicsTree`](https://github.com/robotology/codyco-modules/blob/master/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamicsThread.cpp#L1510)). 

Currently the are two ways of getting the floating base position estimate in the `yarpWholeBodyState` interface: using the "internal" method implemented in [1] in which  a single link (specified in the configuration file) is always assume to be fixed or the "external" method implemented in [2] in which the 
floating base related information are read from an external port. 
The only module that at the moment publish the information needed by the external port is the `wholeBodyDynamicsTree` module, using its own internal `singleLeggedOdometry` class. Porting this logic (as described in https://github.com/robotology/idyntree/issues/78) to the `wholeBodyEstimator` would be a good initial exercise from my point of view. 

[1] : https://github.com/robotology/yarp-wholebodyinterface/pull/23
[2] : https://github.com/robotology/yarp-wholebodyinterface/pull/33
[3] :

jeljaik, 2015-10-13:
Thanks @traversaro, perfectly understood. I'll try to port this to the module then.

traversaro, 2015-10-13:
A major problem is how to pass the information "a given link is in contact" to the estimation without starting to open a specific port for each estimator, that is not a scalable approach. 

My current idea is to have a generic "events:i` port in the module that reads a vector of strings that are generic "events". This list of events is then passed to each estimation module, that can then react to the events. We will then have the events `e_l_foot_in_contact` , `e_l_foot_not_in_contact` generated by the controller to inform that estimation in which link is in contact. The cons of this idea is that using unstructured strings is inefficient and relatively dangerous (the sender and the receiver have to agree on the semantics on the events).

jeljaik, 2015-10-13:
@traversaro I'm trying to design the overall structure of the module using StartUML. I think it might be a good way to effectively collaborate on this matter as many components are in place. I'm sketching both a class diagram and a sequence diagram for the matter. Within tomorrow I will show you some initial draft to some extend extrapolated from `quaternionEKF`. I will attach the file to this comment and ping you and @naveenoid right after. Bear with me...

jeljaik, 2015-10-15:
After some talks with @traversaro and @francesco-romano this will be the initial structure of the module. Any suggestion will be well received. I will put an SVG version of this module in the `doc` directory of `wholeBodyEstimator`. It is still missing a couple of things such as the details on the `wholeBodySensors`. I will update this issue as soon as a first skeleton will be pushed to `codyco-modules`. No later than this week.

[image]

jeljaik, 2015-10-18:
## Notes on Task (T06) ##
1. A new class called `LeggedOdometry` inherits from class `IEstimator` to implement the (so far) three basic methods of any estimator namely: `init`, `run`, `release`. 
2. (personal note) A few details on how this porting is being done in https://github.com/robotology/codyco-modules/commit/2a43a9242d7ac2b32c8985154384ecdf18da76ff#diff-e92fb94b541bf74ff3a9c1f0b5d8e06cR195

jeljaik, 2015-10-22:
@traversaro Want me to merge in `master` the current status of the module given that it's stable enough at the moment regarding `LeggedOdometry`?

traversaro, 2015-10-22:
Good idea. We have still to pay attention if we want to add the quaternionEKF stuff and thus depend on bfl, that is not integrated in the superbuild. One way to deal with it is to transform the estimators in proper yarp devices (that expose the IEstimator interface ) so that they can be loaded at runtime ... 
Otherwise some #IFDEF are also ok. :D

jeljaik, 2015-10-22:
Exactly, my point was to merge at this stage after you removed the dependencies on bfl, etc. However I'm not getting quite well the problem with bfl. I believe it is integrated with codyco-superbuild. I'm not compiling BFL outside codyco-superbuild. Can you explain a bit more please?

jeljaik, 2015-10-22:
`simpleLeggedOdometry` has been officially moved from `wholeBodyDynamicsTree` to the `idyntree-estimation` library and implemented in the new module `wholeBodyEstimator` as of 879152bf5fdb64e9c41e337b754e1b3b0842c270

traversaro, 2017-08-10:
Closing for cleanup of old issues.

closed 2017-08-10 by traversaro