robotology-legacy/codyco-modules #134: torqueBalancing segfaulting on `start` command if codycoCommons is compiled in Debug

traversaro, 2015-10-10:


francesco-romano, 2015-10-11:
I think it is related to https://github.com/robotology/codyco-superbuild/issues/85

Unfortunately I cannot replicate on my laptop

traversaro, 2015-10-11:
Ahahaha sorry, yesterday I was wondering why I was not able to find the issue that I had just wrote! Closing that one.

traversaro, 2015-10-11:
You recompiled both codyco-modules and codyco-commons in Debug? I guess otherwise it could also depend on Eigen versions.

francesco-romano, 2015-10-11:
If Xcode did what I asked it yes.
My eigen version is 3.2.6

francesco-romano, 2015-10-21:
@traversaro  Any news on this?

traversaro, 2015-10-21:
I did not further investigated the root of this problem.

traversaro, 2019-03-15:
The `torqueBalancing` module has not been used since a long time, I think it just make sense to close all the issues related to it.

closed 2019-03-15 by traversaro