robotology-legacy/codyco-modules #135: torqueBalancing not working on my laptop

traversaro, 2015-10-10:
torqueBalancing is not able to keep the robot standing on my laptop, even with clock synchronization. On the other hand, `torqueBalancing.mdl` is working fine. It is just a local issue of my pc or torqueBalancing on Gazebo has some problems? 
cc @francesco-romano @DanielePucci

francesco-romano, 2015-10-11:
It has some oscillations (not damped by the way) but it is able to stand for some time.

DanielePucci, 2015-10-11:
I think that the problem resides in the different gains used by the `torqueBalancing.mdl` and `torqueBalancing.c++` (I know, it's a bad notation, but I liked it) when `YARP_ROBOT_NAME` is `icubGazeboSim`. As a matter of fact, the gains used by the former are:

https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/torqueBalancing/robots/icubGazeboSim/gains.m#L19

~~~
    gain.PCOM                 = diag([ 50   50  50]);
    gain.ICOM                 = diag([  0    0   0]);
    gain.DCOM                 = 2*sqrt(gain.PCOM);
    % Impadances acting in the null space of the desired contact forces 

    impTorso            = [   60    60   10
                               0     0    0]; 
    impArms             = [8    8    8   12   
                            0   0    0    0];
                        
    impLeftLeg          = [ 35   20    30     350    550   0
                             0    0     0       0      0   0]; 

    impRightLeg         = [35   20    30      350    550   0
                            0    0     0        0      0   0]; 
~~~

while the gains used by the latter are

https://github.com/robotology/codyco-modules/blob/master/src/modules/torqueBalancing/app/robots/icubGazeboSim/torqueBalancing.ini

~~~
comKp       (50 50 50)
comKd       (0 0 0)
comKi       (0 0 0)

kw          1
kImp        (20 20 10    20 20 20 20   20 20 20 20   30 30 30 60 10 10      30 30 30 60 10 10 )
~~~

Besides the differences in the `gain.impedances`, the main point is the lack of dumping at the CoM level, as @francesco-romano was actually saying. Notice that the `gain.DCOM = 2*sqrt(gain.PCOM)` ensures that the eigenvalues are all real (and negative) which helps minimize the oscillations. Sincerely, I don't know the reason why this difference. May you try to make them equal and see if these oscillations still arise?

francesco-romano, 2015-11-14:
I noticed yesterday that the kd gains for the gazebo models were zero. I updated the gains.
Can you try again @traversaro ?

traversaro, 2017-08-10:
Bug lost in time, closing.

closed 2017-08-10 by traversaro