robotology-legacy/codyco-modules #136: [torqueBalancing] Gain scheduling for impedances and damping

DanielePucci, 2015-10-11:
We need to implement a gain scheduling policy for the gains associated with the torque balancing control. How shall we proceed?

iron76, 2015-10-11:
It is quite a delicate issue since it might be eventually implemented in the firmware. Therefore, you should discuss the implementation with @randaz81.

randaz81, 2015-10-12:
ok for implementing (basic) gain scheduling for joint level torque control at fw level (already discussed with @DanielePucci). However if this issue relates to **high-level** stuff (stiffness/damping in the cartesian space / COM control **during balancing**, as the title seems to suggest), there nothing that can be done at the moment.

DanielePucci, 2015-10-12:
Yes @randaz81, I think that @iron76 has confused the low-level gain scheduling policy (that is anyway really important) with the high-level one, which was the main subject of this issue. We may open another issue related to the low-level gain scheduling.

DanielePucci, 2015-10-12:
Since the RPC interface is already ready for gain scheduling in torqueBalancing, I close this issue. Only tests must be carried on.

closed 2015-10-12 by DanielePucci