robotology-legacy/codyco-modules #141: wholeBodyDynamicsTree always use the `l_foot` as the world frame in iCubGui

traversaro, 2015-10-23:
See the use of `left_foot_link_idyntree_id` variable in https://github.com/robotology/codyco-modules/blob/master/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamicsThread.cpp#L1293 . 

@hu-yue I cannot work on this until Sunday, but it should be quite easy to fix adding an option to wholeBodyDynamicsTree, so feel free to try to solve it if you want.

hu-yue, 2015-10-26:
@traversaro Ok, I've looked into the code, but I'm not 100% sure that I'm getting the point.

This ```if``` https://github.com/robotology/codyco-modules/blob/master/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamicsThread.cpp#L1312 here is the one that causes the iCubGui to visualize everything wrt the left foot as I've understood, since by changing it to right_foot now everything is wrt right foot.
But how to adjust this so that it always takes the right frame as reference?

hu-yue, 2015-10-26:
@traversaro from the code I see that in the code you use ```r_foot``` and ```l_foot```, while in my configuration files I've always used ```l_sole``` or ```r_sole```, so should they be ```_foot```??

traversaro, 2015-10-26:
@hu-yue apparently the use of `l_foot` was a leftover from the past... in practice it is equivalent to use `r_foot` or one of its rigidly connected frames (`r_sole`,`r_foot_dh_frame`).

Can you try the fix in https://github.com/robotology/codyco-modules/tree/fix/issue141 ? Thanks.

hu-yue, 2015-10-27:
@traversaro just tried it, it switches the reference frame when using ```--assume_fixed``` 
So as far as I've understood the workaround now is that it takes the frame in ```--assume_fixed``` as reference and if it is not specified it used the ```l_foot```, right?

traversaro, 2015-10-27:
Yes. Basically it is not really relevant if it uses `l_foot` or `l_sole` because the transformation sent to the `iCubGui` is just `rootLinkAsStartup_H_rootLink` , and the fixed link is just used because it assumed to be in a constant position.

traversaro, 2015-10-27:
Merging if it is working.

hu-yue, 2015-10-27:
Yes it is working thanks for the fix!

traversaro, 2015-11-28:
Still a lot of work to do to cleanup the wholeBodyDynamicsTree and what it publishes to the iCubGui, but the specific bug is fixed, so we can close the issue.

closed 2015-11-28 by traversaro