robotology-legacy/codyco-modules #146: [T02] wholeBodySensors interface should be configured to read the sensors from URDF

jeljaik, 2015-10-26:
The wholeBodySensors interface should be configured to read the sensors present in the URDF file. Once we have a working URDF --> iDynTree::SensorsList parser, this boils down to iterate on elements of the iDynTree::SensorsList and add their names on wbi::idLists that are then used to configure the wholeBodySensors . The work to do here is relative to the iDynTree and yarpWholeBodyInterface libraries:
- [ ] `iDynTree`, parsing sensors information from URDF : robotology/idyntree#30
- [ ] `yarpWholeBodyInterface` , reading accelerometer, gyroscope and orientation information as published on eth iCub : robotology/yarp-wholebodyinterface#42

jeljaik, 2015-11-16:
@naveenoid Can you please update this issue? You told me verbally about the commits but since we're accurately keeping track of this module through a milestone, I'd love to have everything referenced and checked here please :) Also, if you can provide us with a detailed explanation on how things work now, what's left and so on, I would be grateful. If you consider there are big issues pending, please create one with the proper label and add it to the wholeBodyEstimator milestone.

naveenoid, 2015-11-16:
The sequence of commits are towards this PR : https://github.com/robotology/idyntree/pull/89

In this PR I address URDF reading of accelerometers and gyroscopes. I then build and integrate the URDF parsed sensors into a sensorList class. I will be writing some note shortly but basically new sensors are included in this form  (for eg.) :  

    <sensor name="dummy_accl" type="accelerometer">
        <parent type="link">link1</parent>
    </sensor>
    <sensor name="dummy_gyro" type="gyroscope">
        <parent type="link">link1</parent>
        <pose xyz="0 0 0" rpy="0 -0 0" />
    </sensor>

The next commit scheduled for today / tomorrow will demonstrate it working on a simple test. Will document that here. 

The sequence of tasks are highlighted below 
- [ ] Parser debugging for incorrect tags (no crashing should happen)
- [ ] Parser memory leak fixing
- [ ] Finalise Sensor class interface after documentation and debate
- [ ] to determine frame in which Sensor Measurements must be obtained
- [ ] In Accelerometer, debate if measurement should be made in link frame (i.e. measurement of link acceleration coming in from accelerometer, and link spatial velocity)
- [ ] Integrate F/T sensor with the <sensor> tag style (currently they are implemented using <gazebo> tags
- [ ] Extract positions of MTB on iCub from python script
- [ ] Start incorporating gyroscopes and accelerometers on iCub URDF

If you think this whole sequence needs a new issue, I can proceed with it and document the results here when done.

jeljaik, 2015-11-16:
:heart_eyes_cat:  Bravo!

naveenoid, 2015-11-17:
UnitTestURDFGenericSensorImport presents first working tests. Checkout sample URDF in https://github.com/robotology/idyntree/blob/feature/sensor_expansion/tests/data/twoLinksFixed.urdf and https://github.com/robotology/idyntree/blob/feature/sensor_expansion/tests/data/oneLink.urdf

jeljaik, 2015-11-25:
Navee, [this](https://github.com/robotology/codyco-modules/blob/newModule/wholeBodyEstimator/src/modules/wholeBodyEstimator/src/QuaternionEKF.cpp#L284-L335) is the piece of code I would like to avoid through `yarpWholeBodySensors`. Let me know if it's unclear.

traversaro, 2015-11-25:
@jeljaik @naveenoid My two cents on how that code should be implemented in the `yarpWholeBodySensors` : https://github.com/robotology/yarp-wholebodyinterface/issues/42#issuecomment-159638184 .

jeljaik, 2016-02-04:
@naveenoid news?

naveenoid, 2016-02-06:
Current status  being updated https://github.com/robotology/codyco-modules/issues/147#issuecomment-180067857 

I will keep refining list as components get defined and finished.

traversaro, 2017-08-10:
Closing for cleanup of old issues.

closed 2017-08-10 by traversaro