robotology-legacy/codyco-modules #156: [torqueBalancing] load PID gains from config - howto

hu-yue, 2015-11-13:
Hi, 
I wanted to try to load the gains from the config file as per #150 , however I understood from this sentence in the readme:
```
The original gains read from the firmware will be overwritten with the specified gains in the 
configuration file (it is then possible to specify only the gains that must change).
```
that in the config file where I store the PID gains I can store only those that I actually want to change instead of all the joints, e.g. a file ```newGains.ini``` that looks like:
```
torso_roll = ("kp",-85),("ki",0),("kd",0)
torso_pitch = ("kp",-85),("ki",0),("kd",0)
r_ankle_roll = ("kp",-200),("ki",0),("kd",0)
```
But what happened is that these were the only joints with torque PIDs set, while all the other joints had ```0```.
Is this the expected behavior and I just misinterpreted the instructions?
Thanks!

francesco-romano, 2015-11-13:
Hi @hu-yue.
You are perfectly right in your interpretation. 
So probably there is a bug somewhere.

Just a question, before proceeding with further investigation, you are using the real robot, aren't you?
Gazebo does not support torque PIDs.

hu-yue, 2015-11-13:
@francesco-romano yes with the real robot.
I launched the torqueBalancing module and then the robot fell and the controller stopped due to joint limits, then I checked the gains and saw that they were as described before. Then I stopped the controller with "stop" in the rpc and the original gains were restored.

hu-yue, 2015-11-13:
Stopped in the sense that it went back to position control mode.

francesco-romano, 2015-11-13:
@hu-yue can you try now?
If this does not work, please reopen the issue :smile:

hu-yue, 2015-11-13:
Actually now they are not set to ```0``` but it seems that it's just not taking the gains at all, since they did not change from the default values.
I don't have the rights to reopen the issue.

hu-yue, 2015-11-13:
Also, a question, when are the new gains loaded? When I I start the ```torqueBalancing``` or when I do ```start``` from rpc? I checked the gains from the ```yarpmotorgui```

francesco-romano, 2015-11-13:
At module startup

francesco-romano, 2015-11-13:
Ok.. I tried them on gazebo and they work.

The issue (I think) is that you are passing the gains as "integer". Unfortunately the yarp `Value` class thinks that an integer value is not a double. 
Now, I can "complicate" a bit the code to handle also this case, but I think a better solution is to correct the behaviour directly in yarp ( @lornat75  what do you think? )

@hu-yue  if adding `.0` to all the gains solves the issue you can close.

cc @traversaro

hu-yue, 2015-11-16:
It works adding .0 after the number, for us it's ok to keep it as it is for now.

hu-yue, 2015-11-23:
Btw @francesco-romano, do you have plans on loading also the other parameters from configuration files? (e.g. ```kbemf ```)

francesco-romano, 2015-11-23:
I'll be very happy to do that.
Unfortunately it seems the `Pid` class does not have this parameter.
@randaz81  can we add it?

closed 2015-11-23 by francesco-romano