robotology-legacy/codyco-modules #157: Decide on a design patter for better scalability

jeljaik, 2015-11-13:
As it has been discussed, there is a basic interface for all the estimators instantiated by this module, however more than one estimator needs to be instantiated at a particular time. In addition to this, we want to specify the desired estimators through configuration file and have the flexibility to add more derived classes from `IEstimator` without impacting the source code. After a bit of research I found out that this correspond to a creational design pattern called `Factory Pattern (with Registry)`. A first attempt to exemplify what I wanna implement is shown [here](https://github.com/jeljaik/snippets/tree/master/IEstimatorsFactoryTest). Please check it out @traversaro. The piece of code is compilable and I like the idea, hope you too. This is simulating the map of classes read from config file as we had previously discussed by typing it in terminal.

traversaro, 2015-11-13:
:+1: That is exactly. the idea that i had in mind on the long period. I was actually imaging to use the yarp devices machinery to get automatically the support for estimators loadable at runtime, but also this approach is ok for me.

jeljaik, 2015-11-13:
Awesome! I spent all day figuring it out lol. I'll port it to the real classes and push the changes :)

jeljaik, 2015-11-16:
First implementation done in: 248d98eed63e5a360a3037d38775d184410a783c

closed 2015-11-16 by jeljaik