robotology-legacy/codyco-modules #158: Move 'torqueBalancing' and 'wholeBodyDynamicsTree' to separate repository

francesco-romano, 2015-11-14:
And integrate them in the superbuild.

Given their always-increasing use on the robot for demos and as a basic for experiments I think these two modules deserve to live in a separate repository (so that they can be installed without all the other codyco-modules stuff).

What do you think? @traversaro @DanielePucci @iron76

traversaro, 2015-11-17:
I actually think that we should cleanup this repo, and probably all will remain will be just `torqueBalancing`, `wholeBodyDynamicsTree` and few other modules, that there will be no need to separate further. Perhaps anyway it would make sense to reset this repository to get rid of all the history in the repo, that given that this was the original `codyco` repository is quite huge.

traversaro, 2019-06-09:
Time has passed, so I updated the issue title considering that:
* `wholeBodyDynamicsTree` YARP module has been deprecated, and its functionality are provided by the `wholebodydynamics` YARP device
* `torqueBalancing` YARP module has been deprecated, and its functionality are now provided by the controllers provided in 

So, the remaining thing to move in a new repo is just  the `wholebodydynamics` device. As dicussed in https://github.com/robotology/whole-body-estimators/pull/1#issuecomment-487022806, the device will be moved to https://github.com/robotology/whole-body-estimators .

traversaro, 2020-02-01:
Fixed by https://github.com/robotology/whole-body-estimators/pull/16 .

traversaro, 2020-03-23:
https://github.com/robotology/whole-body-estimators/pull/16 is now closed, we can close this issue.

closed 2020-03-23 by traversaro