robotology-legacy/codyco-modules #159: WBD3 Segfault on exit

francesco-romano, 2015-11-20:
~~~
#0  0x000000000048a140 in wholeBodyDynamicsThread::closePort (this=0xb3f000, _port=0x7ffff7dd21d0 <vtable for yarp::os::PortWriter+16>)
    at /home/icub/software/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamicsThread.cpp:1954
#1  0x0000000000485792 in wholeBodyDynamicsThread::closeOdometry (this=0xb3f000)
    at /home/icub/software/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamicsThread.cpp:1139
#2  0x0000000000489f48 in wholeBodyDynamicsThread::threadRelease (this=0xb3f000)
    at /home/icub/software/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamicsThread.cpp:1944
#3  0x00007ffff7b67d25 in theExecutiveBranch(void*) () from /home/icub/software/yarp/build/lib/libYARP_OS.so.1
#4  0x00007ffff599fb16 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.0.3.so
#5  0x00007ffff549c182 in start_thread (arg=0x7ffee8ff9700) at pthread_create.c:312
#6  0x00007ffff5d27fbd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
~~~

francesco-romano, 2015-11-23:
This does not happen on `iCubGenova02`

hu-yue, 2015-11-23:
It happend for iCubHeidelberg01 today.
I think before I did not notice it because I used to launch it from the manager, today I launched it from terminal and on exit it gave Segmentation fault.

francesco-romano, 2015-11-23:
Does it happen every time?

hu-yue, 2015-11-23:
I'm not sure, I have to check. I launched WBD3 only once from the terminal today. Can try that again.

hu-yue, 2015-11-23:
Yes it happens everytime, just tried a couple of runs and it always exits with Segfault.

hu-yue, 2015-11-23:
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff63fff700 (LWP 19234)]
0x000000000042c2a9 in wholeBodyDynamicsThread::closePort(yarp::os::Contactable*) ()
(gdb) bt
#0  0x000000000042c2a9 in wholeBodyDynamicsThread::closePort(yarp::os::Contactable*) ()
#1  0x000000000042c32f in wholeBodyDynamicsThread::closeOdometry() ()
#2  0x0000000000431642 in wholeBodyDynamicsThread::threadRelease() ()
#3  0x00007ffff7b67a95 in theExecutiveBranch(void*) () from /usr/local/src/robot/yarp/build-x86_64/lib/libYARP_OS.so.1
#4  0x00007ffff599cb16 in ACE_OS_Thread_Adapter::invoke() () from /usr/lib/libACE-6.0.3.so
#5  0x00007ffff5499182 in start_thread (arg=0x7fff63fff700) at pthread_create.c:312
#6  0x00007ffff5d2447d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) 
```

traversaro, 2015-12-03:
I am unable to reproduce this issue.. : [ 
The configuration files for `yarpWholeBodyInterface` and `wholeBodyDynamicsTree` are the one in the repositories?

traversaro, 2015-12-03:
This https://github.com/robotology/codyco-modules/issues/159 should fix it.

francesco-romano, 2015-12-04:
@hu-yue can you test this?

@traversaro maybe you wanted to write "This https://github.com/robotology/codyco-modules/pull/163 should fix it" ?

traversaro, 2015-12-04:
Yes!

hu-yue, 2015-12-04:
@francesco-romano @traversaro the fix does fix the segfault issue, however, I had problems with ```torqueBalancing``` on right leg. The left leg fell despite all the parameters are correctly set. In fact, going back to the master branch of ```codyco-modules``` the leg did not fall.

francesco-romano, 2015-12-04:
So probably this is an issue with the base.
Can you paste here the configuration of the balancing and of the yarpWhole etc .ini file? (of this second file, you can avoid pasting the joint list.. I'm curious about the configuration of the base

hu-yue, 2015-12-04:
balancing

```
name        torqueBalancing
robot       icub
period      10
modulePeriod 0.01

wbi_config_file yarpWholeBodyInterface.ini
wbi_joint_list ROBOT_TORQUE_CONTROL_JOINTS

constraint_links ("r_sole")

comKp       (40 47 40)
comKd       (0 0 0)
comKi       (0 0 0)

kw          1
kImp        (100 150 100    30  30 30 30 40 40    100 70 65 30 20 20)

tsat        ( 24  24  24    24  24 24 24 24 24    24  24 24 24 24 24)

torqueGains torqueGainsRightStanding.ini
```
yarpWholeBodyInterface first rows
```
robot          icub
urdf           model.urdf
getLimitsFromControlBoard

[WBI_STATE_OPTIONS]
localWorldReferenceFrame r_sole
estimateBaseState
```

Don't know if needed, wholeBodyDynamicsTree first rows
```
name            wholeBodyDynamicsTree
period          10
robot           icub
cutoff          3.0

#This file is referenced here, but it
# should be found in a robot specific directory
wbi_conf_file yarpWholeBodyInterface.ini
```

Btw, these are local files, they are not the same as in the repository ever since we changed the torqueBalancing on single leg from left to right leg.

francesco-romano, 2015-12-04:
Thanks @hu-yue 
Let's wait for @traversaro thoughts. 
If needed we can also Skype/Hangouts

traversaro, 2015-12-04:
There is too much cruft that accumulated related to `base link`/calibration stuff. 
Doing a bit of open loop cleaning is probably the most efficient way of understanding what is going on, I will come back to you to test again after a first round of cleaning.

traversaro, 2015-12-04:
In the day I am busy with TU/e related tasks, and this weekend I can't work on it. 
Any fix is delayed until one of the next week evenings. 

@hu-yue are you ok to switch to master and keep the old behavior for the moment ?

hu-yue, 2015-12-04:
@traversaro totally fine, it's not urgent for us

traversaro, 2017-07-28:
Won't fix, wholeBodyDynamicsTree will be deprecated soon as descriebed in https://github.com/robotology/codyco-modules/issues/223 .

closed 2017-07-28 by traversaro