robotology-legacy/codyco-modules #160: Bad estimation of external wrenches: are the measurements from FT reliable?

DanielePucci, 2015-11-25:
When performing the yoga demo on the left foot starting from a two-feet standing position, the algorithms in charge of estimating the external wrenches sense external forces even if there is no external perturbation exerted on the robot:

[image]

The algorithms estimating the external wrenches basically evaluate the "difference" between the robot's model and the measurements coming from the FT sensors to identify an external wrench (@traversaro can add additional and precise details on this). Now, thanks to the work at the modelling level by @fiorisi and @Mick3Lozzo, the model of iCubV2.5 shall be reliable at both kinematics and dynamics level. This means that the focus is on the measurements coming from the FT sensors. Probably,  we need in-situ calibrations of the FT sensors, which may account for platform dependent effect that miscalibrate the FT sensors during the installation procedure.

@francesco-romano @ fiandra @iron76

traversaro, 2020-02-01:
I think this issue can be closed, not because the issue has been  solved, but because the issue of the reliability of the FT sensors measurements is not related at all to the software for estimating the external forces and torques. However, I think that this issue and all the other issues that refer to this one may be of interest for the team that is currently tackling the issue of reliability of FT sensors measurements, including @marcoaccame @Nicogene @triccyx @salvi-mattia @davidetome @simeonedussoni @maurizbo .

traversaro, 2020-02-01:
It may be an interesting reference also for all the @robotology/iit-dic .

closed 2020-02-01 by traversaro