robotology-legacy/codyco-modules #164: Use nicer looking names for wholeBodyEstimator groups

traversaro, 2015-12-04:
Currenty, if you need to set a configuration parameter from the command for a specific estimator, you can do it through the command line with:
```
wholeBodyEstimator --SIMPLE_LEGGED_ODOMETRY::initial_world_frame l_sole 
```
The choice of capital words separated by an underscore make this command rather unreadable . Perhaps 
we should switch to lower case names for the groups? 

cc @jeljaik

jeljaik, 2016-01-11:
Since the decision of using the factory pattern design for this module, the group names in `wholeBodyEstimator.ini` are actually the names of the corresponding classes, i.e. the command line you previously mentioned turns into:
`wholeBodyEstimator --LeggedOdometry::initial_world_Frame`

jeljaik, 2016-01-11:
See for instance the [configuration file](https://github.com/robotology/codyco-modules/blob/newModule/wholeBodyEstimator/src/modules/wholeBodyEstimator/app/robots/iCubGenova02/wholeBodyEstimator.ini) for iCubGenova01

closed 2016-01-11 by jeljaik

jeljaik, 2016-01-11:
Also changed existing lower-case variables [here](https://github.com/robotology/codyco-modules/commit/9407a2913f1771bd07ba8b97b9c73f7f858d95fb)