robotology-legacy/codyco-modules #165: torqueBalancingReferenceGenerator fails to init if torqueBalancing module is running

nunoguedelha, 2015-12-18:
torqueBalancingReferenceGenerator module was run from terminal on Purple iCub. Find below the yarp log:
```
icub@icub14:~/.local/share/yarp/robots/iCubGenova02$ torqueBalancingReferencesGenerator 
||| clearing context
||| adding context [torqueBalancingReferencesGenerator]
||| configuring
||| no policy found
||| default config file specified as torqueBalancingRefGen.ini
||| checking [/home/icub/.local/share/yarp/robots/iCubGenova02/torqueBalancingRefGen.ini] (pwd)
||| found /home/icub/.local/share/yarp/robots/iCubGenova02/torqueBalancingRefGen.ini
||| finding file [wbi_config_file]
||| checking [/home/icub/.local/share/yarp/robots/iCubGenova02/yarpWholeBodyInterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubGenova02] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubGenova02] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubGenova02
||| checking [/etc/yarp/robots/iCubGenova02] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-x86_64/share/yarp/robots/iCubGenova02] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-x86_64/share/iCub/robots/iCubGenova02] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-x86_64/share/iCub/robots/iCubGenova02
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGenova02] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/codyco-superbuild/build-x86_64/install/share/codyco/robots/iCubGenova02] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/codyco-superbuild/build-x86_64/install/share/codyco/robots/iCubGenova02
||| checking [/usr/local/src/robot/icub-tests/suits/robots/iCubGenova02] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-x86_64/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-x86_64/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/codyco-superbuild/build-x86_64/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-tests/suits/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-x86_64/share/iCub/robots/iCubGenova02/yarpWholeBodyInterface.ini] (robot)
||| checking [/usr/local/src/robot/codyco-superbuild/build-x86_64/install/share/codyco/robots/iCubGenova02/yarpWholeBodyInterface.ini] (robot)
||| found /usr/local/src/robot/codyco-superbuild/build-x86_64/install/share/codyco/robots/iCubGenova02/yarpWholeBodyInterface.ini
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /torqueBalancingReferencesGeneratoractuator/torso/rpc:o active at tcp://10.0.0.14:10274
yarp: Port /torqueBalancingReferencesGeneratoractuator/torso/command:o active at tcp://10.0.0.14:10275
yarp: Port /torqueBalancingReferencesGeneratoractuator/torso/state:i active at tcp://10.0.0.14:10276
yarp: Port /torqueBalancingReferencesGeneratoractuator/torso/stateExt:i active at tcp://10.0.0.14:10277
yarp: Sending output from /torqueBalancingReferencesGeneratoractuator/torso/rpc:o to /icub/torso/rpc:i using tcp
yarp: Datagram packet size set to 65000
yarp: Sending output from /torqueBalancingReferencesGeneratoractuator/torso/command:o to /icub/torso/command:i using udp
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/torso/state:o to /torqueBalancingReferencesGeneratoractuator/torso/state:i using udp
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/torso/stateExt:o to /torqueBalancingReferencesGeneratoractuator/torso/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /torqueBalancingReferencesGeneratoractuator/left_arm/rpc:o active at tcp://10.0.0.14:10278
yarp: Port /torqueBalancingReferencesGeneratoractuator/left_arm/command:o active at tcp://10.0.0.14:10279
yarp: Port /torqueBalancingReferencesGeneratoractuator/left_arm/state:i active at tcp://10.0.0.14:10280
yarp: Port /torqueBalancingReferencesGeneratoractuator/left_arm/stateExt:i active at tcp://10.0.0.14:10281
yarp: Sending output from /torqueBalancingReferencesGeneratoractuator/left_arm/rpc:o to /icub/left_arm/rpc:i using tcp
yarp: Datagram packet size set to 65000
yarp: Sending output from /torqueBalancingReferencesGeneratoractuator/left_arm/command:o to /icub/left_arm/command:i using udp
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/left_arm/state:o to /torqueBalancingReferencesGeneratoractuator/left_arm/state:i using udp
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/left_arm/stateExt:o to /torqueBalancingReferencesGeneratoractuator/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /torqueBalancingReferencesGeneratoractuator/right_arm/rpc:o active at tcp://10.0.0.14:10282
yarp: Port /torqueBalancingReferencesGeneratoractuator/right_arm/command:o active at tcp://10.0.0.14:10283
yarp: Port /torqueBalancingReferencesGeneratoractuator/right_arm/state:i active at tcp://10.0.0.14:10284
yarp: Port /torqueBalancingReferencesGeneratoractuator/right_arm/stateExt:i active at tcp://10.0.0.14:10285
yarp: Sending output from /torqueBalancingReferencesGeneratoractuator/right_arm/rpc:o to /icub/right_arm/rpc:i using tcp
yarp: Datagram packet size set to 65000
yarp: Sending output from /torqueBalancingReferencesGeneratoractuator/right_arm/command:o to /icub/right_arm/command:i using udp
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/right_arm/state:o to /torqueBalancingReferencesGeneratoractuator/right_arm/state:i using udp
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/right_arm/stateExt:o to /torqueBalancingReferencesGeneratoractuator/right_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[... 111 lines cut ...]
yarp: Removing input from /icub/left_arm/state:o to /torqueBalancingReferencesGeneratorstate/left_arm/state:i
yarp: Datagram packet size set to 65000
yarp: Removing input from /icub/left_arm/stateExt:o to /torqueBalancingReferencesGeneratorstate/left_arm/stateExt:i
yarp: Removing output from /torqueBalancingReferencesGeneratorstate/right_arm/rpc:o to /icub/right_arm/rpc:i
yarp: Removing output from /torqueBalancingReferencesGeneratorstate/right_arm/command:o to /icub/right_arm/command:i
yarp: Datagram packet size set to 65000
yarp: Removing input from /icub/right_arm/state:o to /torqueBalancingReferencesGeneratorstate/right_arm/state:i
yarp: Datagram packet size set to 65000
yarp: Removing input from /icub/right_arm/stateExt:o to /torqueBalancingReferencesGeneratorstate/right_arm/stateExt:i
yarp: Removing output from /torqueBalancingReferencesGeneratorstate/left_leg/rpc:o to /icub/left_leg/rpc:i
yarp: Removing output from /torqueBalancingReferencesGeneratorstate/left_leg/command:o to /icub/left_leg/command:i
yarp: Datagram packet size set to 65000
yarp: Removing input from /icub/left_leg/state:o to /torqueBalancingReferencesGeneratorstate/left_leg/state:i
yarp: Datagram packet size set to 65000
yarp: Removing input from /icub/left_leg/stateExt:o to /torqueBalancingReferencesGeneratorstate/left_leg/stateExt:i
yarp: Removing output from /torqueBalancingReferencesGeneratorstate/right_leg/rpc:o to /icub/right_leg/rpc:i
yarp: Removing output from /torqueBalancingReferencesGeneratorstate/right_leg/command:o to /icub/right_leg/command:i
yarp: Datagram packet size set to 65000
yarp: Removing input from /icub/right_leg/state:o to /torqueBalancingReferencesGeneratorstate/right_leg/state:i
yarp: Datagram packet size set to 65000
yarp: Removing input from /icub/right_leg/stateExt:o to /torqueBalancingReferencesGeneratorstate/right_leg/stateExt:i
[ERR] Error while initializing yarpWholeBodyStates interface.
Could not initialize wbi object.
RFModule failed to open
```

nunoguedelha, 2015-12-18:
FYI @naveenoid @francesco-romano

traversaro, 2015-12-18:
If the `torqueBalancing` is not running instead `torqueBalancingReferenceGenerator` is working fine?

jeljaik, 2015-12-18:
Yes

traversaro, 2015-12-18:
I think it is related to https://github.com/robotology-playground/WBI-Toolbox-controllers/issues/10 .

naveenoid, 2015-12-21:
From my understanding of yarpWholeBodySensors, it seems to be triggered around this line : 
https://github.com/robotology/yarp-wholebodyinterface/blob/master/src/yarpWholeBodySensors.cpp#L720

This implies its an issue with the IOpenLoopControl interface (i.e. the getOutputs method)

DanielePucci, 2015-12-21:
Update: I get this error when doing a simple getEstimate of joint position

DanielePucci, 2015-12-21:
@francesco-romano

DanielePucci, 2015-12-21:
Update: it does not happen on the CAN robot `iCubGenova03`

traversaro, 2019-03-15:
The torqueBalancing module has not been used since a long time, I think it just make sense to close all the issues related to it.

closed 2019-03-15 by traversaro