robotology-legacy/codyco-modules #166: CTRL_RAD2DEG and CTRL_DEG2RAD scope

rlober, 2016-01-05:
So I just did a fresh install of superbuild and got a scope error related to these two constants. I did a massive find replace 
```
CTRL_RAD2DEG ==> iCub::ctrl::CTRL_RAD2DEG
CTRL_DEG2RAD ==> iCub::ctrl::CTRL_DEG2RAD
```
throughout superbuild (most of the offenders are in `codyco-modules`) and it compiled fine. Haven't run anything yet to check for bugs but I was hoping you guys could come up with a simpler fix. If not I can submit a pull request for the changes.

traversaro, 2016-01-05:
I think most of them should have been fixed in https://github.com/robotology/codyco-modules/commit/60a2ef2da194370219682114df4b70b9d6a9201d .. perhaps you had an outdated checkout of the repo for some reason?

rlober, 2016-01-06:
You know what, you are probably right. I thought the machine I was on had a totally fresh clone of superbuild but I guess not. Sorry about this!

closed 2016-01-06 by rlober

ahoarau, 2016-02-01:
Having the same issue here :(

ahoarau, 2016-02-01:
But fixed using ```make update-all``` !