robotology-legacy/codyco-modules #168: [jointTorqueControl] get/set coulombVelThresh & frictionCompensation

rlober, 2016-02-05:
Is there a way to set the variables [`coulombVelThr`](https://github.com/robotology/codyco-modules/blob/master/src/modules/jointTorqueControl/include/jointTorqueControl/jointTorqueControlThread.h#L93) and [`frictionCompensationFactor`](https://github.com/robotology/codyco-modules/blob/master/src/modules/jointTorqueControl/include/jointTorqueControl/jointTorqueControlThread.h#L120). From the torque control interface `ITorqueControl`?

If not, how would I go about implementing this?

iron76, 2016-02-05:
@randaz81 should know the details. I remember him playing with similar parameter from the robotMotorGui.

rlober, 2016-02-08:
Thank @iron76

randaz81, 2016-02-08:
Hi @rlober, no there is not.
The variables you are mentioning (whose meaning is obscure to me) are specific to the JointTorqueControl module and not to the ITorqueControl yarp interface. What you can do is edit the module to set these parameters interactively through an rpc port. I'd not go for extending the ITorqueControl interface with unnecessary methods...

iron76, 2016-02-08:
I guess these parameters map one to one with similar parameters in the `ITorqueControl` interface. @DanielePucci @traversaro and @randaz81 might help @rlober in finding the correct mapping if such on-to-one or almost-one-to-one mapping exists.

rlober, 2016-02-08:
@randaz81 

Admittedly I am not sure what `coulombVelThresh` does but `frictionCompensation` basically regulates how much the modeled friction is incorporated into the PID output. http://wiki.icub.org/codyco/dox/html/group__codyco__jointTorqueControl.html

Do you have any documentation on the PID computations for the `ITorqueControl` I would like to see how the various parameters are implemented.

@iron76 

That was my thought, that the `setMotorTorqueParameters` behavior could be overloaded when using JTC. But this could make the code nasty behind the scenes. Another option like @randaz81 suggested would be to open an RPC port on JTC for the specific purpose of modifying these two parameters for a single joint. The latter may be the uglier solution but in the long run I think we will do away with JTC so maybe quick and dirty is better.

Are there still discrepancies between Eth and Can robots w.r.t. `ITorqueControl`?

traversaro, 2016-02-08:
The *coulomb velocity threshold* is meant to be exposed in the `embObjMotionControl` using the `IRemoteVariables` interface (even it is not implemented at the moment):
 https://github.com/robotology/icub-main/blob/d1936c07ca0b8299288ebde60941c5e8d63e4e79/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4205

The `IRemoveVariables` interface (as introduced in https://github.com/robotology/yarp/pull/500) is meant to: 
> The advanced developer can use them to set/get internal variables which are not exposed by the remote device with a proper yarp interface. This interface should be used for debug purposes only. To be NOT used in standard applications.

The rationale is that for tuning such implementation-specific parameters it may be enough to have a generic key-value interface instead of overloading the `ITorqueControl` or `IPidControl` interface with multiple implementation-specific parameters.

rlober, 2016-02-08:
@traversaro Thanks for the info. Although I still am unclear as to how I can modify these parameters even with the `IRemoveVariables` interface? 

When I run JTC for the left arm for instance the following ports:
```
registration name /icubGazeboSim/jtc/left_arm/command:i ip 134.157.18.48 port 10045 type tcp
registration name /icubGazeboSim/jtc/left_arm/quit ip 134.157.18.48 port 10048 type tcp
registration name /icubGazeboSim/jtc/left_arm/rpc:i ip 134.157.18.48 port 10044 type tcp
registration name /icubGazeboSim/jtc/left_arm/state:o ip 134.157.18.48 port 10046 type tcp
registration name /icubGazeboSim/jtc/left_arm/stateExt:o ip 134.157.18.48 port 10047 type tcp
registration name /icubGazeboSim/jtc_proxy/left_arm/command:o ip 134.157.18.48 port 10041 type tcp
registration name /icubGazeboSim/jtc_proxy/left_arm/rpc:o ip 134.157.18.48 port 10040 type tcp
registration name /icubGazeboSim/jtc_proxy/left_arm/state:i ip 134.157.18.48 port 10042 type tcp
registration name /icubGazeboSim/jtc_proxy/left_arm/stateExt:i ip 134.157.18.48 port 10043 type tcp
```

Also it looks like the `PARAM_ID_FRICTION_COMPENSATION` and `PARAM_ID_COULOMB_VEL_THR` have been linked to pointers to the vectors containing these variables: [L59](https://github.com/robotology/codyco-modules/blob/master/src/modules/jointTorqueControl/src/jointTorqueControlThread.cpp#L59) & [L64](https://github.com/robotology/codyco-modules/blob/master/src/modules/jointTorqueControl/src/jointTorqueControlThread.cpp#L64).

How do I go through the rpc to set a specific value ?

traversaro, 2016-02-08:
Currently `jointTorqueControl` does not implemented the `IRemoteVariables` interface, but you can implement it and then use the `yarpmotorgui` to set the Remote Variables for the jtc device. 

The `PARAM_ID_FRICTION_COMPENSATION` and `PARAM_ID_COULOMB_VEL_THR` are parameters of the old joint torque control module (as opposed to the jtc device). It should be removed from the repo ( https://github.com/robotology/codyco-modules/issues/58#issuecomment-148908192 ), but removing it is still in my backlog. : (

rlober, 2016-02-08:
I see that I was looking at the wrong code... Anayways looking at `jointTorqueControl.cpp` it looks like the `ITorqueControl` functions are wrapped by JTC? Also I didn't see a wrapper for `set/getMotorTorqueParameters`[here](https://github.com/robotology/yarp/blob/d371c2a4e2ae32f7da74803e9ad033da0c4558b3/src/libYARP_dev/include/yarp/dev/ITorqueControl.h#L124) and [here](https://github.com/robotology/yarp/blob/d371c2a4e2ae32f7da74803e9ad033da0c4558b3/src/libYARP_dev/include/yarp/dev/ITorqueControl.h#L117). Maybe this function could be wrapped and instead of getting or setting say `ktau` and `ktau_scale` we could get/set `coulombVelThr` and `frictionCompensationFactor`. 

What u think?

traversaro, 2016-02-08:
Yes, it would make sense to implement both `set/getMotorTorqueParameters` in JTC, but without modifications to the `yarp::dev::MotorTorqueParameters` class in yarp you can't use that interface to set some other parameters.

traversaro, 2018-05-14:
fyi @lrapetti

rlober, 2018-05-16:
old issue... self destructing in T-minus 10... 9... 8...

closed 2018-05-16 by rlober

traversaro, 2018-05-16:
:D