robotology-legacy/codyco-modules #170: [jointTorqueControl] launch with Gazebo

rlober, 2016-02-08:
How do I launch JTC with Gazebo?

francesco-romano, 2016-02-08:
Do you need it?
Gazebo already implements to torque control

rlober, 2016-02-08:
Hey @francesco-romano 

I want to test JTC specific PID param setting/getting. #168 

Just for testing, not for actual control.

francesco-romano, 2016-02-08:
Ok. That was the second option :smiley: 

Have you tried something?
Like copying and pasting the configuration on the yarpmanager for the robot?
There already exist the configuration files for gazebo (even if I think the coupling should be removed).

rlober, 2016-02-08:
As of now, I haven't tried anything yet. I actually am not sure how the Gazebo plugins load `.ini` files - if at all?

francesco-romano, 2016-02-08:
You can launch it as a module. Take a look on how it is launched on the real robot.

rlober, 2016-02-08:
Thanks fra I will give it a shot and let you know

rlober, 2016-02-08:
Hey man. Looks like I just had to launch the module as you said there are even `.ini` files already made for `icubGazeboSim`. 

For future reference.
```
codycomoddev --from jtc_[part_name].ini
```

thanks @francesco-romano

closed 2016-02-08 by rlober

francesco-romano, 2016-02-08:
Great!
One thing I noticed is that there are coupling matrices in the .ini file.
To test set/get they are not an issue, but otherwise they should be removed I think