robotology-legacy/codyco-modules #173: Weird warnins

DanielePucci, 2016-02-19:


DanielePucci, 2016-02-19:
~~~
[WARNING]yarpWholeBodyStates: you are using the depreacted estimateBasePosAndVel parameter.
[WARNING]                     please use the new estimateBaseState parameter instead.
[WARN] yarpWholeBodyStates::loadCouplingsFromConfigurationFile : WBI_YARP_JOINTS_MOTOR_KINEMATIC_COUPLINGS group not found in configuration file
~~~

francesco-romano, 2016-02-19:
Well..
The first one is not weird. 
It is the second one that is always present.

closed 2016-05-16 by traversaro

traversaro, 2016-05-16:
Fixed in https://github.com/robotology/yarp-wholebodyinterface/commit/b5035d7f7409983532a408a53755cb3f90aad611 and https://github.com/robotology/yarp-wholebodyinterface/commit/884d6abea0fa8825be6c69b8daf19700505493fa .