robotology-legacy/codyco-modules #177: [torqueBalancing] Find a good position for the icub attached to the mechanical crane

traversaro, 2016-03-19:
The position should be such that lowering the crane the iCub should be able to be on two feet without any human assistance. 

Technically speaking, such a property can be stated geometrically in the following way. Let's define the "sole plane" as the shared plane of the two soles, and the 
"vertical plane" as the plane defined by there points: the robot COM, and the two points in the shoulder at which the robot is connected to the crane. In the desirable configuration the vertical plane and the sole plane should be perpendicular.

traversaro, 2019-03-15:
This is still a relevant issue I think, what do you think @gabrielenava ? :D

gabrielenava, 2019-03-15:
Guess the issue is slightly outdated. In fact, now the robot does not even need the crane anymore:

[image]

gabrielenava, 2019-03-23:
Closing as outdated issue. Home position for iCub + crane has been already selected and it is now pretty stable.

closed 2019-03-23 by gabrielenava