robotology-legacy/codyco-modules #183: yarpWholebodyModel.h not found

hu-yue, 2016-05-09:
Hi,
I was updating everything after some time, but got this error on codyco-modules while compiling the superbuild:

```
Scanning dependencies of target codycoTrajGenDemoY2
[ 53%] Building CXX object src/modules/codycoTrajGenDemoY2/CMakeFiles/codycoTrajGenDemoY2.dir/src/main.cpp.o
[ 54%] Building CXX object src/modules/codycoTrajGenDemoY2/CMakeFiles/codycoTrajGenDemoY2.dir/src/Coordinator.cpp.o
In file included from /usr/local/src/robot/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyEstimator/include/QuaternionEKF.h:28:0,
                 from /usr/local/src/robot/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyEstimator/include/WholeBodyEstimatorThread.h:41,
                 from /usr/local/src/robot/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyEstimator/include/WholeBodyEstimatorModule.h:26,
                 from /usr/local/src/robot/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyEstimator/src/main.cpp:7:
/usr/local/src/robot/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyEstimator/include/floatingBase.h:9:55: fatal error: yarpWholeBodyInterface/yarpWholebodyModel.h: No such file or directory
 #include <yarpWholeBodyInterface/yarpWholebodyModel.h>
                                                       ^
compilation terminated.
[ 56%] Building CXX object src/modules/walkPlayer/CMakeFiles/walkPlayer.dir/src/main.cpp.o
[ 58%] Building CXX object src/modules/walkPlayer/CMakeFiles/walkPlayer.dir/src/scriptModule.cpp.o
[ 59%] Building CXX object src/modules/walkPlayer/CMakeFiles/walkPlayer.dir/src/actionClass.cpp.o
/usr/local/src/robot/codyco-superbuild/main/codyco-modules/src/modules/walkPlayer/src/scriptModule.cpp: In member function ‘virtual bool scriptModule::configure(yarp::os::ResourceFinder&)’:
/usr/local/src/robot/codyco-superbuild/main/codyco-modules/src/modules/walkPlayer/src/scriptModule.cpp:25:35: warning: ‘yarp::os::ConstString yarp::os::ResourceFinder::getContextPath()’ is deprecated (declared at /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/ResourceFinder.h:248) [-Wdeprecated-declarations]
     contextPath=rf.getContextPath().c_str();
                                   ^
make[5]: *** [src/modules/wholeBodyEstimator/CMakeFiles/wholeBodyEstimator.dir/src/main.cpp.o] Error 1
make[4]: *** [src/modules/wholeBodyEstimator/CMakeFiles/wholeBodyEstimator.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
Linking CXX executable ../../../bin/walkPlayer
Linking CXX executable ../../../bin/codycoTrajGenDemoY2
[ 59%] Built target walkPlayer
[ 59%] Built target codycoTrajGenDemoY2
make[3]: *** [all] Error 2
make[2]: *** [main/codyco-modules/CMakeFiles/YCMStamp/codyco-modules-build] Error 2
make[1]: *** [CMakeFiles/codyco-modules.dir/all] Error 2
make: *** [all] Error 2
```

Everything is on master branch updated this morning.
Thanks!

@traversaro @jeljaik

traversaro, 2016-05-09:
I guess `CODYCO_USES_OROCOS_BFL_BERDY` is set to on, right?

hu-yue, 2016-05-09:
Yes, but when I changed it to off nothing seemed to have changed, I'll try to clean the cache and build again.

hu-yue, 2016-05-09:
@traversaro ok now it's compiling fine. 
But how come is this happening?
Anyway I think I don't need the `BFL_BERDY`, as this was because some time ago I was using Jorh's module but now it should not be needed anymore.

closed 2016-05-09 by hu-yue

traversaro, 2016-05-09:
I think there was this option that remained in your cache.