robotology-legacy/codyco-modules #194: Fix iCubGui visualization during YOGA++ demo

traversaro, 2016-06-30:
The `iCubGui` visualization is shitty during the `YOGA++` demo, for a number of reason: 
* For robot models generated using the v2 generation workflow directly from Creo model, the frames used by the iCubGui for visualizing contact forces are not present (except for `l_foot_dh_frame` and `r_foot_dh_frame`) : https://github.com/robotology-playground/icub-model-generator/issues/11 . 
* The `iCubGui` has a long standing bug for which forces under the floor are not correctly visualized : https://github.com/robotology/icub-main/issues/29 . We typically used the workaround of visualizing the robot levitating 1 meter over the ground to overcome this issue. 
* The `wholeBodyDynamicsDevice` is not publishing the floating base position estimate, as the old `wholeBodyDynamicsTree` did. The rationale is that logic could be moved to another device specifically designed for odometry, that could as well be loaded in the `pc104`.  The class implementing the computational part is already implemented in http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1SimpleLeggedOdometry.html#a189c13e9ff5d64f3b8b05034f7cedb64 , we just need to write the yarp device (probably recycling some of the code from the `wholeBodyDynamicsDevice`).

iron76, 2016-07-01:
A nice visualisation is fundamental for presenting our work to daily visitors. Do we have an estimate of how much effort is needed for this issue to be addressed?

traversaro, 2016-08-16:
The floating base estimator bit should be fixed by https://github.com/robotology/codyco-modules/pull/197 . For the DH frame stuff, it would take some more time, probably two/three days.