robotology-legacy/codyco-modules #196: [Floating Base Estimator] reset to Arbitrary Frame

KevinAStein, 2016-08-10:
Hi,
I try to use the floating base estimator and need to reset it at the beginning. Therefore I take the floating base values from my inverse kinematics and use 

```resetSimpleLeggedOdometryToArbitraryFrame("root_link", floating_base_transform, "root_link");```

In the transform I have a rotation Matrix and a translation. After the reset the angles I get from the yarp port /floatingBaseEstimator/base:o match the ones I used to construct the Rotation Matrix, while the translation does not.

I figured that it works if I first multiply the translation with the rotation Matrix and put these values in the transformation.

Is it supposed to work like this?
thanks

@hu-yue @traversaro

traversaro, 2016-08-10:
Hi @KevinAStein , 
  the `initial_reference_frame_H_world` argument specified in https://github.com/robotology/codyco-modules/blob/improvFloatingBase/src/idl/floatingBaseEstimatorRPC/codyco/floatingBaseEstimatorRPC.thrift#L57 is indeed not fully defined in the function header, I am sorry. : ( 

Typically when I write A_H_B in the code it is a short hand for the homegeneous transform that transforms a vector expressed in the `B` frame to the `A` frame (this is equivalent to ${}^A H_B$ as defined in this document : http://repository.tue.nl/849895 ) .

Consecutively if you get from your IK the initial world_H_root_link  , you have to pass to the estimator root_link_H_world, i.e. its inverse.

KevinAStein, 2016-08-12:
ok, thanks for the explanation.

closed 2016-08-12 by KevinAStein