robotology-legacy/codyco-modules #198: wholeBodyDynamicsTree with JTC

jeljaik, 2016-08-30:
Hello guys,

I am trying to launch `wholeBodyDynamicsTree` (WBDT) using a new URDF model of `iCubParis02` result of a merge between the upper body params of `iCubGenova05` and ours. The resulting URDF can be found [here](https://github.com/robotology/yarp-wholebodyinterface/blob/master/app/robots/iCubParis02/model_merged.urdf). I'm launching WBDT with the following parameters: 

```
wholeBodyDynamicsTree --autoconnect --from wholeBodyDynamicsTree_jtc.ini --assume_fixed root_link
```

Where [wholeBodyDynamicsTree_jtc.ini](https://github.com/robotology/codyco-modules/blob/master/src/modules/wholeBodyDynamicsTree/app/robots/iCubParis02/wholeBodyDynamicsTree_jtc.ini) specifies the use of [yarpWholeBodyInterface_mergedURDF.ini](https://github.com/robotology/yarp-wholebodyinterface/blob/master/app/robots/iCubParis02/yarpWholeBodyInterface_mergedURDF.ini) and none of links in the list `[IDYNTREE_SKINDYNLIB_LINKS]` has the `*dh_frame` ending as in the previous URDF. In turn, `yarpWholeBodyInterface` uses [model_merged.urdf](https://github.com/robotology/yarp-wholebodyinterface/blob/master/app/robots/iCubParis02/model.urdf).

The error I then get is the following: 
```
macbook:iCubParis02 jorhabibeljaik$ wholeBodyDynamicsTree --autoconnect --verbose --from wholeBodyDynamicsTree_jtc.ini --assume_fixed root_link
||| clearing context
||| adding context [wholeBodyDynamicsTree]
||| configuring
||| default config file specified as wholeBodyDynamicsTree_jtc.ini
||| checking [/Users/jorhabibeljaik/Code/codyco-superbuild/build/install/share/codyco/robots/iCubParis02/wholeBodyDynamicsTree_jtc.ini] (pwd)
||| found /Users/jorhabibeljaik/Code/codyco-superbuild/build/install/share/codyco/robots/iCubParis02/wholeBodyDynamicsTree_jtc.ini
[INFO]assume_fixed option found, using  root_link  as fixed link as a kinematic root instead of the imu.
yarp: Port /wholeBodyDynamicsTree/rpc:i active at tcp://10.0.0.106:10371
||| finding file [wbi_conf_file]
||| checking [/Users/jorhabibeljaik/Code/codyco-superbuild/build/install/share/codyco/robots/iCubParis02/yarpWholeBodyInterface_mergedURDF.ini] (pwd)
||| found /Users/jorhabibeljaik/Code/codyco-superbuild/build/install/share/codyco/robots/iCubParis02/yarpWholeBodyInterface_mergedURDF.ini
[INFO] yarpWholeBodySensors: initializing with 32 encoders 0 pwm sensors 6 F/T sensors 1 IMUs
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/jtc/torso/rpc:o active at tcp://10.0.0.106:10943
yarp: Port /wholeBodyDynamicsTree/jtc/torso/command:o active at tcp://10.0.0.106:10944
yarp: Port /wholeBodyDynamicsTree/jtc/torso/state:i active at tcp://10.0.0.106:10945
yarp: Port /wholeBodyDynamicsTree/jtc/torso/stateExt:i active at tcp://10.0.0.106:10946
yarp: Sending output from /wholeBodyDynamicsTree/jtc/torso/rpc:o to /icub/jtc/torso/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamicsTree/jtc/torso/command:o to /icub/jtc/torso/command:i using udp
yarp: Receiving input from /icub/jtc/torso/state:o to /wholeBodyDynamicsTree/jtc/torso/state:i using udp
yarp: Receiving input from /icub/jtc/torso/stateExt:o to /wholeBodyDynamicsTree/jtc/torso/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/head/rpc:o active at tcp://10.0.0.106:10376
yarp: Port /wholeBodyDynamicsTree/head/command:o active at tcp://10.0.0.106:10377
yarp: Port /wholeBodyDynamicsTree/head/state:i active at tcp://10.0.0.106:10378
yarp: Port /wholeBodyDynamicsTree/head/stateExt:i active at tcp://10.0.0.106:10379
yarp: Sending output from /wholeBodyDynamicsTree/head/rpc:o to /icub/head/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamicsTree/head/command:o to /icub/head/command:i using udp
yarp: Receiving input from /icub/head/state:o to /wholeBodyDynamicsTree/head/state:i using udp
yarp: Receiving input from /icub/head/stateExt:o to /wholeBodyDynamicsTree/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/jtc/left_arm/rpc:o active at tcp://10.0.0.106:10963
yarp: Port /wholeBodyDynamicsTree/jtc/left_arm/command:o active at tcp://10.0.0.106:10964
yarp: Port /wholeBodyDynamicsTree/jtc/left_arm/state:i active at tcp://10.0.0.106:10965
yarp: Port /wholeBodyDynamicsTree/jtc/left_arm/stateExt:i active at tcp://10.0.0.106:10966
yarp: Sending output from /wholeBodyDynamicsTree/jtc/left_arm/rpc:o to /icub/jtc/left_arm/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamicsTree/jtc/left_arm/command:o to /icub/jtc/left_arm/command:i using udp
yarp: Receiving input from /icub/jtc/left_arm/state:o to /wholeBodyDynamicsTree/jtc/left_arm/state:i using udp
yarp: Receiving input from /icub/jtc/left_arm/stateExt:o to /wholeBodyDynamicsTree/jtc/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/jtc/right_arm/rpc:o active at tcp://10.0.0.106:10967
yarp: Port /wholeBodyDynamicsTree/jtc/right_arm/command:o active at tcp://10.0.0.106:10968
yarp: Port /wholeBodyDynamicsTree/jtc/right_arm/state:i active at tcp://10.0.0.106:10969
yarp: Port /wholeBodyDynamicsTree/jtc/right_arm/stateExt:i active at tcp://10.0.0.106:10970
yarp: Sending output from /wholeBodyDynamicsTree/jtc/right_arm/rpc:o to /icub/jtc/right_arm/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamicsTree/jtc/right_arm/command:o to /icub/jtc/right_arm/command:i using udp
yarp: Receiving input from /icub/jtc/right_arm/state:o to /wholeBodyDynamicsTree/jtc/right_arm/state:i using udp
yarp: Receiving input from /icub/jtc/right_arm/stateExt:o to /wholeBodyDynamicsTree/jtc/right_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/jtc/left_leg/rpc:o active at tcp://10.0.0.106:10971
yarp: Port /wholeBodyDynamicsTree/jtc/left_leg/command:o active at tcp://10.0.0.106:10972
yarp: Port /wholeBodyDynamicsTree/jtc/left_leg/state:i active at tcp://10.0.0.106:10973
yarp: Port /wholeBodyDynamicsTree/jtc/left_leg/stateExt:i active at tcp://10.0.0.106:10974
[... 24 lines cut ...]
yarp: Receiving input from /icub/right_leg/analog:o to /wholeBodyDynamicsTree/ftSens/r_leg_ft_sensor:i using udp
yarp: Port /wholeBodyDynamicsTree/ftSens/r_foot_ft_sensor:i active at tcp://10.0.0.106:10395
yarp: Receiving input from /icub/right_foot/analog:o to /wholeBodyDynamicsTree/ftSens/r_foot_ft_sensor:i using udp
yarp: Port /wholeBodyDynamicsTree/imu/imu_frame:i active at tcp://10.0.0.106:10396
yarp: Receiving input from /icub/inertial to /wholeBodyDynamicsTree/imu/imu_frame:i using udp
[INFO]Launching wholeBodyDynamicsThread with name :  wholeBodyDynamicsTree  and robotName  icub  and period  10
||| finding file model_merged.urdf
||| checking [/Users/jorhabibeljaik/Code/codyco-superbuild/build/install/share/codyco/robots/iCubParis02/model_merged.urdf] (pwd)
||| found /Users/jorhabibeljaik/Code/codyco-superbuild/build/install/share/codyco/robots/iCubParis02/model_merged.urdf
urdfdom: setting link 'base_link' materialurdfdom: successfully added a new link 'base_link'urdfdom: setting link 'root_link' materialurdfdom: link 'root_link' material 'green' defined in Visual.urdfdom: successfully added a new link 'root_link'urdfdom: setting link 'r_hip_1' materialurdfdom: link 'r_hip_1' material 'material.metal' defined in Visual.urdfdom: successfully added a new link 'r_hip_1'urdfdom: setting link 'r_hip_2' materialurdfdom: setting link 'r_hip_2' material to 'material.metal'urdfdom: successfully added a new link 'r_hip_2'urdfdom: setting link 'r_hip_3' materialurdfdom: successfully added a new link 'r_hip_3'urdfdom: setting link 'r_upper_leg' materialurdfdom: setting link 'r_upper_leg' material to 'material.metal'urdfdom: successfully added a new link 'r_upper_leg'urdfdom: setting link 'r_lower_leg' materialurdfdom: setting link 'r_lower_leg' material to 'material.metal'urdfdom: successfully added a new link 'r_lower_leg'urdfdom: setting link 'r_ankl [... cut ...]  100
8   wholeBodyDynamicsTree               0x000000010d7a749a _ZN23wholeBodyDynamicsModule9configureERN4yarp2os14ResourceFinderE + 17690
9   libYARP_OS.1.dylib                  0x000000010fd359a3 _ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE + 51
10  wholeBodyDynamicsTree               0x000000010d798054 main + 5332
11  libdyld.dylib                       0x00007fff8e6b85ad start + 1
12  ???                                 0x0000000000000007 0x0 + 7
```

Thanks for the time in advance.

francesco-romano, 2016-08-30:
Maybe I'm missing something, but why WBD3 should know about the presence of the JTC?
@traversaro ?

jeljaik, 2016-08-30:
Fixed in 90dbede0f662c81d0ba825880b34d54ca38c5041

closed 2016-08-30 by jeljaik