robotology-legacy/codyco-modules #200: Get external wrenches estimation to work on the Walkman robot

traversaro, 2016-09-10:
@gabrielenava has been recently struggling to get the estimation of external wrenches to work on the Walkman robot, using the old KDL-based `wholeBodyDynamicsTree`. He got stuck on several points in which the code was hardcoding iCub-specific information. 

The last bug of this kind was the hardcoding of the `imu_frame` name as the frame in which the IMU outputs its measurements. While it could be possible to update the old `wholeBodyDynamicsTree`, I think it would be a waste of resources. The new `wholebodydynamics` device can be easily run outside the main `yarprobotinterface` program of the robot using the new module `wholeBodyDynamics3` (pun intended), so I think it is easier to just put effort in improving the new software. 
The `wholeBodyDynamics3` is a sort of little `yarprobotinterface` (actually in the future it could be replaced with a dedicated configuration of the `yarprobotinterface`), i.e. it launches several YARP devices and connects them appropriately. An example configuration file for `icubGazeboSim` is https://github.com/robotology/codyco-modules/tree/master/src/modules/wholeBodyDynamics3/app/robots/icubGazeboSim . The only problem is that at the moment `wholebodydynamics` to run needs some features only available in the `devel` branch of YARP, but I think we can start to work on it with @gabrielenava and wait for the release of YARP 2.3.68 to switch to `wholeBodyDynamics3` for all the codyco users.

traversaro, 2016-09-10:
I am going through the code to see if there some robot-specific leftovers (beside https://github.com/robotology/codyco-modules/issues/199).

traversaro, 2016-09-11:
At the moment this issue is blocking: https://github.com/robotology/idyntree/issues/230 .

traversaro, 2018-06-04:
> actually in the future it could be replaced with a dedicated configuration of the yarprobotinterface

Note that this happened.

gabrielenava, 2019-01-07:
Being the activity on Walkman not carried on at the moment, maybe we can close this issue. Also, I think this issue is now outdated.

traversaro, 2019-01-07:
I agree.

closed 2019-01-07 by traversaro