robotology-legacy/codyco-modules #202: homePoseBalancingTwoFeet.ini is assymetric for iCubGenova01, iCubGenova02 and iCubDarmstadt01

traversaro, 2016-09-20:
Example from https://github.com/robotology/codyco-modules/blob/master/src/modules/torqueBalancing/app/robots/iCubGenova02/homePoseBalancingTwoFeet.ini : 
~~~
[right_leg_zero]             
// ...
PositionZero                12            5             0            -10           -4            -5        
 
[left_leg_zero]              
// ...
PositionZero                12            5             0            -10           -3            -5       
~~~

traversaro, 2016-09-20:
Source of the problem (propagated through copy pasting) : https://github.com/robotology/codyco-modules/commit/6d0eb90a1d4499e755462a4f89f804dea53c4446

traversaro, 2016-09-20:
cc @DanielePucci @gabrielenava

traversaro, 2016-09-20:
For reference: in `iCubGenova02`, the current poses assumes a relative positions of the foot given by: 
~~~
l_sole_H_r_sole:

ans =

    0.9997    0.0197   -0.0174   -0.0012
   -0.0197    0.9998   -0.0009   -0.2155
    0.0174    0.0013    0.9998   -0.0001
         0         0         0    1.0000
~~~
That can be computed by the following matlab script: 
~~~matlab
mdlLoader = iDynTree.ModelLoader();

% Load model with just legs 
consideredJoints = iDynTree.StringVector();
consideredJoints.push_back('l_hip_pitch');
consideredJoints.push_back('l_hip_roll');
consideredJoints.push_back('l_hip_yaw');
consideredJoints.push_back('l_knee');
consideredJoints.push_back('l_ankle_pitch');
consideredJoints.push_back('l_ankle_roll');
consideredJoints.push_back('r_hip_pitch');
consideredJoints.push_back('r_hip_roll');
consideredJoints.push_back('r_hip_yaw');
consideredJoints.push_back('r_knee');
consideredJoints.push_back('r_ankle_pitch');
consideredJoints.push_back('r_ankle_roll');

mdlLoader.loadReducedModelFromFile('../data/iCubGenova02.urdf',consideredJoints);

kinDynMdl = iDynTree.KinDynComputations();
kinDynMdl.loadRobotModel(mdlLoader.model());

% Consideter joint positions : homePosTwoFeetBalancing
% see https://github.com/robotology/codyco-modules/issues/202
dq = iDynTree.JointDOFsDoubleArray(mdlLoader.model());
dq.zero();
gravity = iDynTree.Vector3();
gravity.zero();

q = iDynTree.JointPosDoubleArray(mdlLoader.model());
qMat = q.toMatlab();
% left leg 
qMat(1) = deg2rad(12.0);
qMat(2) = deg2rad(5.0);
qMat(3) = deg2rad(0.0);
qMat(4) = deg2rad(-10.0);
qMat(5) = deg2rad(-3.0);
qMat(6) = deg2rad(-5.0);
% right leg 
qMat(7) = deg2rad(12.0);
qMat(8) = deg2rad(5.0);
qMat(9) = deg2rad(0.0);
qMat(10) = deg2rad(-10.0);
qMat(11) = deg2rad(-4.0);
qMat(12) = deg2rad(-5.0);

q.fromMatlab(qMat);

% Set state 
kinDynMdl.setRobotState(q,dq,gravity);

% Compute l_sole_H_r_sole 
fprintf('l_sole_H_r_sole:\n');
kinDynMdl.getRelativeTransform('l_sole','r_sole').asHomogeneousTransform().toMatlab()
~~~

DanielePucci, 2016-09-26:
thanks @traversaro

gabrielenava, 2018-09-03:
In order to close this very old issue, I made symmetric all the `homePositions` for all robots with this [PR](https://github.com/robotology/codyco-modules/pull/269). For the robot balancing on two feet, there were only differences of 1-2 [deg] between the left and right joints on the legs (in particular, the ankles). I imagine these differences have been used to compensate for encoder calibration errors on the real robot, but this is not the correct way to deal with them. I think we can close this issue after merging the PR. Tested in simulaion with `icubGazeboSim` and `iCubGazeboV2_5`.

closed 2018-09-10 by gabrielenava