robotology-legacy/codyco-modules #204: Compilation of devices not working with yarp devel

traversaro, 2016-09-22:


traversaro, 2016-09-22:
cc @gabrielenava

gabrielenava, 2016-09-22:
After the last update to the branch devel in yarp, I tried to compile codyco-superbuild. I got the following error:

```
error: ‘PassThroughControlBoard’ was not declared in this scope
 YARP_DEFINE_SHARED_SUBCLASS(passThroughControlBoard, PassThroughControlBoard, yarp::dev::DeviceDriver)
                                                      ^
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: note: suggested alternative:
In file included from /home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:10:0:
/home/gnava/Software/codyco-superbuild/main/codyco-modules/src/devices/jointTorqueControl/PassThroughControlBoard.h:22:18: note:   ‘yarp::dev::PassThroughControlBoard’
 class yarp::dev::PassThroughControlBoard :  public DeviceDriver,
                  ^
In file included from /home/gnava/Software/Install/include/yarp/os/SharedLibraryFactory.h:13:0,
                 from /home/gnava/Software/Install/include/yarp/os/SharedLibraryClassFactory.h:10,
                 from /home/gnava/Software/Install/include/yarp/os/SharedLibraryClass.h:10,
                 from /home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:8:
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:1: error: ‘cn’ was not declared in this scope
 YARP_DEFINE_SHARED_SUBCLASS(passThroughControlBoard, PassThroughControlBoard, yarp::dev::DeviceDriver)
 ^
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: error: expected type-specifier before ‘PassThroughControlBoard’
 YARP_DEFINE_SHARED_SUBCLASS(passThroughControlBoard, PassThroughControlBoard, yarp::dev::DeviceDriver)
                                                      ^
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: error: expected ‘;’ before ‘PassThroughControlBoard’
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:1: error: cannot dynamic_cast ‘cn’ (of type ‘<type error>’) to type ‘class yarp::dev::DeviceDriver*’ (source is not a pointer)
 YARP_DEFINE_SHARED_SUBCLASS(passThroughControlBoard, PassThroughControlBoard, yarp::dev::DeviceDriver)
 ^
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:1: error: type ‘<type error>’ argument given to ‘delete’, expected pointer
 YARP_DEFINE_SHARED_SUBCLASS(passThroughControlBoard, PassThroughControlBoard, yarp::dev::DeviceDriver)
 ^
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp: In function ‘void passThroughControlBoard_destroy(void*)’:
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: error: expected type-specifier before ‘PassThroughControlBoard’
 YARP_DEFINE_SHARED_SUBCLASS(passThroughControlBoard, PassThroughControlBoard, yarp::dev::DeviceDriver)
                                                      ^
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: error: expected ‘>’ before ‘PassThroughControlBoard’
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: error: expected ‘(’ before ‘PassThroughControlBoard’
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: error: ‘PassThroughControlBoard’ was not declared in this scope
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:54: note: suggested alternative:
In file included from /home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:10:0:
/home/gnava/Software/codyco-superbuild/main/codyco-modules/src/devices/jointTorqueControl/PassThroughControlBoard.h:22:18: note:   ‘yarp::dev::PassThroughControlBoard’
 class yarp::dev::PassThroughControlBoard :  public DeviceDriver,
                  ^
In file included from /home/gnava/Software/Install/include/yarp/os/SharedLibraryFactory.h:13:0,
                 from /home/gnava/Software/Install/include/yarp/os/SharedLibraryClassFactory.h:10,
                 from /home/gnava/Software/Install/include/yarp/os/SharedLibraryClass.h:10,
                 from /home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:8:
/home/gnava/Software/codyco-superbuild/build/main/codyco-modules/src/devices/jointTorqueControl/yarp_plugin_codycomod_passThroughControlBoard.cpp:30:1: error: expected primary-expression before ‘>’ token
[... cut ...]
 YARP_DEFINE_SHARED_SUBCLASS(passThroughControlBoard, PassThroughControlBoard, yarp::dev::DeviceDriver)
 ^
make[5]: *** [src/devices/jointTorqueControl/CMakeFiles/jointTorqueControl.dir/yarp_plugin_codycomod_passThroughControlBoard.cpp.o] Error 1
make[4]: *** [src/devices/jointTorqueControl/CMakeFiles/jointTorqueControl.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [main/codyco-modules/CMakeFiles/YCMStamp/codyco-modules-build] Error 2
make[1]: *** [CMakeFiles/codyco-modules.dir/all] Error 2
make: *** [all] Error 2

```

traversaro, 2016-09-22:
Fixed by https://github.com/robotology/codyco-modules/commit/af6d573b1052dcc854ac7fbdc135b9ff9599ac69 .

closed 2016-09-22 by traversaro