robotology-legacy/codyco-modules #205: ENABLE_codycomod_* needs to be explicitly set to ON to compile a device

traversaro, 2016-09-22:
Using YARP 2.3.67 we can set this option explicitly to ON since https://github.com/robotology/yarp/pull/895 .

drdanz, 2016-09-26:
@traversaro I'm about to add an `INTERNAL` option that will set DEFAULT=ON and hide the option from the CMake gui.
Also, since I noticed you used it in b0c4f0b, I'm also changing `ADVANCED ON` in just `ADVANCED` as I don't see a reason for setting `ADVANCED OFF`, sorry for the mess...

drdanz, 2016-09-26:
@traversaro `INTERNAL` option added... You can ensure that is available by bumping yarp dependency to `2.3.67.2`

traversaro, 2016-09-26:
@drdanz At the moment I prefer not to bump the `codyco-superbuild` to YARP `devel`. For the moment I keep compatibility with YARP 2.3.66, with the only difference that if you use YARP 2.3.66 you have to manually enable the devices.

traversaro, 2016-09-26:
@drdanz I think this doc line needs to be updated : https://github.com/robotology/yarp/blob/devel/conf/YarpPlugin.cmake#L118 .

traversaro, 2016-09-26:
Working if YARP is 2.3.67.2 , otherwise the user still needs to manually set to ON the `ENABLE_` variables.

closed 2016-09-26 by traversaro