robotology-legacy/codyco-modules #207: Can't launch module with icubGazeboSim.

jeljaik, 2016-10-04:
Hi guys,

I wanted to do some tests using `SimpleLeggedOdometry`, therefore was trying first with the old way done in `wholeBodyDynamicsTree`. 

The following: 
`wholeBodyDynamicsTree --robot icubGazeboSim --asume_fixed_from_odometry l_sole`

Gives me this error: 
`[ERR] yarpWholeBodySensors::init() error: failing in opening force/torque sensors`

I also have my `YARP_ROBOT_NAME` variable set correctly to the simulator in case you wonder.

traversaro, 2016-10-04:
Mhh.. Can you provide the complete log of `wholeBodyDynamicsTree` ?

jeljaik, 2016-10-04:
Yup, sorry. 

```bash
jorhabib@jorh-codyco:~/code/icub-gazebo/world$ YARP_ROBOT_NAME=icubGazeboSim wholeBodyDynamicsTree --robot icubGazeboSim
||| clearing context
||| adding context [wholeBodyDynamicsTree]
||| configuring
||| default config file specified as wholeBodyDynamicsTree.ini
||| checking [/home/jorhabib/code/icub-gazebo/world/wholeBodyDynamicsTree.ini] (pwd)
||| checking [/home/jorhabib/.config/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_HOME)
||| checking [/home/jorhabib/.local/share/yarp/robots/icubGazeboSim] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_DIRS)
||| checking [robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| checking [/home/jorhabib/code/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| found /home/jorhabib/code/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim
||| checking [/usr/local/share/codyco/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| checking [config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/jorhabib/code/codyco-superbuild/build/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/jorhabib/code/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/wholeBodyDynamicsTree.ini] (robot)
||| found /home/jorhabib/code/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/wholeBodyDynamicsTree.ini
yarp: Port /wholeBodyDynamicsTree/rpc:i active at tcp://134.157.18.79:10376
||| finding file [wbi_conf_file]
||| checking [/home/jorhabib/code/icub-gazebo/world/yarpWholeBodyInterface.ini] (pwd)
||| checking [/home/jorhabib/code/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini] (robot)
||| found /home/jorhabib/code/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/torso/rpc:o active at tcp://134.157.18.79:10377
yarp: Port /wholeBodyDynamicsTree/torso/command:o active at tcp://134.157.18.79:10378
yarp: Port /wholeBodyDynamicsTree/torso/state:i active at tcp://134.157.18.79:10379
yarp: Port /wholeBodyDynamicsTree/torso/stateExt:i active at tcp://134.157.18.79:10380
yarp: Sending output from /wholeBodyDynamicsTree/torso/rpc:o to /icubGazeboSim/torso/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamicsTree/torso/command:o to /icubGazeboSim/torso/command:i using udp
yarp: Receiving input from /icubGazeboSim/torso/state:o to /wholeBodyDynamicsTree/torso/state:i using udp
yarp: Receiving input from /icubGazeboSim/torso/stateExt:o to /wholeBodyDynamicsTree/torso/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/head/rpc:o active at tcp://134.157.18.79:10381
yarp: Port /wholeBodyDynamicsTree/head/command:o active at tcp://134.157.18.79:10382
yarp: Port /wholeBodyDynamicsTree/head/state:i active at tcp://134.157.18.79:10383
yarp: Port /wholeBodyDynamicsTree/head/stateExt:i active at tcp://134.157.18.79:10384
yarp: Sending output from /wholeBodyDynamicsTree/head/rpc:o to /icubGazeboSim/head/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamicsTree/head/command:o to /icubGazeboSim/head/command:i using udp
yarp: Receiving input from /icubGazeboSim/head/state:o to /wholeBodyDynamicsTree/head/state:i using udp
yarp: Receiving input from /icubGazeboSim/head/stateExt:o to /wholeBodyDynamicsTree/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/left_arm/rpc:o active at tcp://134.157.18.79:10385
yarp: Port /wholeBodyDynamicsTree/left_arm/command:o active at tcp://134.157.18.79:10386
yarp: Port /wholeBodyDynamicsTree/left_arm/state:i active at tcp://134.157.18.79:10387
yarp: Port /wholeBodyDynamicsTree/left_arm/stateExt:i active at tcp://134.157.18.79:10388
yarp: Sending output from /wholeBodyDynamicsTree/left_arm/rpc:o to /icubGazeboSim/left_arm/rpc:i using tcp
yarp: Sending output from /wholeBodyDynamicsTree/left_arm/command:o to /icubGazeboSim/left_arm/command:i using udp
yarp: Receiving input from /icubGazeboSim/left_arm/state:o to /wholeBodyDynamicsTree/left_arm/state:i using udp
yarp: Receiving input from /icubGazeboSim/left_arm/stateExt:o to /wholeBodyDynamicsTree/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[INFO]RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/right_arm/rpc:o active at tcp://134.157.18.79:10389
yarp: Port /wholeBodyDynamicsTree/right_arm/command:o active at tcp://134.157.18.79:10390
yarp: Port /wholeBodyDynamicsTree/right_arm/state:i active at tcp://134.157.18.79:10391
yarp: Port /wholeBodyDynamicsTree/right_arm/stateExt:i active at tcp://134.157.18.79:10392
[... cut ...]
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
yarp: Port /wholeBodyDynamicsTree/ftSens/l_arm_ft_sensor:i active at tcp://134.157.18.79:10401
yarpWholeBodySensors::openFTsens():  /icubGazeboSim/left_arm/analog:o does not exist 
[ERR] yarpWholeBodySensors::init() error: failing in opening force/torque sensors.
[ERROR]wholeBodyDynamicsTree : Error while initializing whole body estimator interface.Closing module 
RFModule failed to open
```

traversaro, 2016-10-04:
I guess `/icubGazeboSim/left_arm/analog:o` exists, right?

jeljaik, 2016-10-04:
Never mind, I had a bad update. Everything's back to normal 👍

closed 2016-10-04 by jeljaik