robotology-legacy/codyco-modules #209: F/T problems while Torque Balancing

paraschos, 2016-10-13:
I am trying to start the torque balancing but the robot falls down (does not apply enough torque
on the legs) and  we press the emergency stop to save the day. Checking the logs, I can see
several F/T sensors in saturation mode:

1673	229.804170	ERROR	 from BOARD 10.0.1.6 (EB6), src CAN2, adr 13, time 352s 773m 206u: (code 0x01000002, par16 0x0004 par64 0x0000000800000000) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .

1676	231.967200	ERROR	 from BOARD 10.0.1.7 (EB7), src CAN2, adr 1, time 354s 933m 263u: (code 0x01000002, par16 0x0000 par64 0x0000000000000026) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .

1684	233.063814	ERROR	 from BOARD 10.0.1.9 (EB9), src CAN2, adr 1, time 356s 19m 545u: (code 0x01000002, par16 0x0000 par64 0x0000000000000043) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .

Any ideas? I recently did a complete firmware update of EMS+CAN of all boards of the robot. Maybe we should recalibrate the sensors? According to the iCub GUI the forces after running the twoFeetStandingIdleAndCalib.sh look good. 

We start `wholeBodyDynamicsTree` with parameters `--autoconnect --name wholeBodyDynamicsTree --robot icub --assume_fixed l_foot_dh_frame` and `torqueBalancing`
with `--from torqueBalancingTwoFeet.ini  --autostart false --check_limits true`

Afterwards we calibrate and we start the torqueBalancing over rpc (normally I do that with autostart to true but just for debugging)

PS: It appears that I am not a member here and therefore I can't apply any labels

paraschos, 2016-10-14:
...or attach a file

paraschos, 2016-10-20:
@DanielePucci any suggestions?

DanielePucci, 2016-10-20:
Hi @paraschos, sorry for taking so long to reply. So, that's kind of weird since the whole left leg and right foot sensors are saturating. In this case, we have to change the offsets of the sensors' ADC through `canLoader --calib`, but I'm not sure that these saturations cause the problem. Can you check if saturations occur after you start `torqueBalancing` or just before? Thanks 

(P.S. Nice move to @cc people :-) so we get direct notifications and this speed things up)

traversaro, 2016-10-20:
Sorry @paraschos , I saw this during IROS and I did not reply immediately and then I forgot to reply, sorry that caused a delay in the reply.

paraschos, 2016-10-20:
Hey @DanielePucci it is a bit strange indeed and I was also wandering if it is the cause or the a symptom of that behavior. I checked the logs (including the torqueBalancing log) and I didn't see anything. Only that it reached a joint limit and the something like "I give up". That occurred (according to the log timestamp) after the saturation.  

The saturation does not occur when we place the robot on the floor. (I will check it again, but pretty sure). Maybe I will also push it a bit to check it, as it could be close to the limit. 

As I wrote above, I updated all the robot boards to the latest firmware but to my understanding the eeprom should be intact. However, if the firmware guys changes anything on the adc function then the calibration limits will be different for sure.

I will play a bit with the offsets then and come back. 

PS: Np Silvio :)

DanielePucci, 2016-10-20:
Mmmmm, I see. Can you launch the `torqueBalancing` with the additional option `--check_limits false`? By doing so, it does not stop the controller when reaching the joint limits.

traversaro, 2016-10-20:
@paraschos You should have the permissions.

paraschos, 2016-10-20:
@DanielePucci Did that, didn't help.
@traversaro ty :)

I am attaching the full log so that you guys can take a look. I will play a bit with the saturation limits. 

[ypmrprunlog_20_10_2016_05_26_01_balance_limits_false.log.txt](https://github.com/robotology/codyco-modules/files/542109/ypmrprunlog_20_10_2016_05_26_01_balance_limits_false.log.txt)

DanielePucci, 2016-10-20:
Mmmmmmmm roger. Do you have the chance to run the Simulink module for the torqueBalancing?

paraschos, 2016-10-20:
I will try it tomorrow. However it also happened once without running the balancing demo.

Log files here @ https://github.com/robotology/icub-support/issues/317

paraschos, 2016-10-20:
I think that I just have to recalibrate the F/Ts. I got more saturation msgs while I started pushing the robot. However I am not sure how to interpreter the amount of messages that I am getting. Are these "a lot" or "a few".

Details of the log collection: First 2 errors (3035/3040) when I placed the robot on the floor for the first time. I lifted and placed the robot down to the floor a couple of times but I didn't get any more errors. The rest of the saturation occurred when I was pushing the robot a bit, mainly in the saggital plane. Communication issues towards the end of the log, where I asked for support in 
https://github.com/robotology/icub-support/issues/317. 

[ypmrprunlog_20_10_2016_08_28_09_pushing_test.log.txt](https://github.com/robotology/codyco-modules/files/542611/ypmrprunlog_20_10_2016_08_28_09_pushing_test.log.txt)

DanielePucci, 2016-10-21:
HI @paraschos, so getting some saturations when the robot is in position control and the user pushes it is quite normal. On the other hand, we do not have a tool to evaluate if the number of saturations you get are fishy. Anyways, we may do the following things (it is up to you to do either or both of them)

1. Shoot a video of the experiment, by taking also the Logger when saturations occur;

2. Re-calibrate the FTs offsets. If you go for this, we will guide you through the procedure.

paraschos, 2016-10-21:
Hi @DanielePucci, when the saturation occurred I was pushing the robot hard enough to make the sole to loose contact. So I guess it is not that fishy at all to have this amount of saturation errors. I am then even more puzzled with the old TorqueBalancing is not working. The last commit of that module was done in Feb...

I will try the simlulink approach first. Maybe we can see something more in that one.

DanielePucci, 2016-10-21:
Roger. Standing by.

paraschos, 2016-10-21:
Hey @DanielePucci, so things started to work. The YOGA demo from the simulink torqueBalancing controller is running (took me some time as I reinstalled all code from scratch recently). However, there is probably a broken eth cable between eb2 and eb3 that is giving a lot of random errors (either link ones or lost packages). 

- There are also a few saturation errors, around 10 per trial. Shall I recalibrate? (I am also attaching 
the logs if you guys want to take a look)

- There is a very small bug(?) in the configuration for our robot. Is looking for the `wholeBodyDynamics` instead of the `wholeBodyDynamicsTree`. I can solve it by passing the `-name` parameter in the `wholeBoDyDynamicsTree` module but I think it is better to add it in the configuration file. I  created a merge request: https://github.com/robotology-playground/WBI-Toolbox-controllers/pull/34

- I tried afterwards to run the cpp `torqueBalancing` but again the robot started to lean forwards
and was loosing height fast. Then I hit the `e-shop`.

- Small documentation fix `CODYCO_USES_WBI_TOOLBOX_CONTROLLERS` should be turned 
to on and not `CODYCO_USES_WBI_TOOLBOX` 

Failed as the crane was not low enough:
[ypmrprunlog_21_10_2016_04_40_20_semi_succ_balance.log.txt](https://github.com/robotology/codyco-modules/files/544711/ypmrprunlog_21_10_2016_04_40_20_semi_succ_balance.log.txt)

Failed due to lost balance:
[ypmrprunlog_21_10_2016_04_40_20_semi_succ_balance2.log.txt](https://github.com/robotology/codyco-modules/files/544713/ypmrprunlog_21_10_2016_04_40_20_semi_succ_balance2.log.txt)

Successful:
[ypmrprunlog_21_10_2016_04_59_19_yoga_suc.log.txt](https://github.com/robotology/codyco-modules/files/544712/ypmrprunlog_21_10_2016_04_59_19_yoga_suc.log.txt)

paraschos, 2016-10-24:
Hey guys,

the networking communication problems are fixed. The YOGA++ works nicely on the robot.

However, I tried the cpp version of the torque balancing on the sim and it fails. Can you take a look? Can I replicate this mode from the simulink? I tried to run the simulink in  `COORDINATOR` mode but matlab crashed. 

Also, do you think that the f/t should be re-calibrated? I think that they look fine...

traversaro, 2020-02-01:
Unfortunatly the C++-based torque balancing controller module has not been mantained, so it is not possible to use it anymore.

closed 2020-02-01 by traversaro