robotology-legacy/codyco-modules #210: Cannot connect to rpc

hu-yue, 2016-10-25:
After updating everything to the devel branch (yarp, icub-main, firmware) in order to use the floating base estimation, we are now not able to connect to the rpc of `wholeBodyDynamicsTree` anymore.
When doing `yarp rpc /wholeBodyDynamicsTree/rpc:i` nothing happens and it seems that the connection is not being established.

@traversaro

traversaro, 2016-10-25:
Mh... Have you recompiled the `codyco-superbuild` and in particular `codyco-modules` after the massive update to the `devel` branch?

hu-yue, 2016-10-25:
Yes, and just to be sure I just did it again, and I still have the same issue. But we can connect using another computer plugged in the network (and on which we did no update anything)

hu-yue, 2016-10-25:
Uhm ok I think it's a stupid issue, sorry, the thing is that before when you open rpc there was `>>` as waiting for the input command, now it's not there and it seemed nothing was happening... I just tried to type in commands and it's working.

closed 2016-10-25 by hu-yue