robotology-legacy/codyco-modules #222: wholeBodyDynamicsDevice verbose unnecessary error

traversaro, 2017-02-22:
~~~
797	12,528212	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
798	12,528240	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
799	12,528272	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
800	12,528299	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
801	12,528325	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material blue not found in model database.
802	12,528350	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material yellow not found in model database.
803	12,528378	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material pink not found in model database.
804	12,528406	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material cyan not found in model database.
805	12,528433	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
806	12,528458	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material white not found in model database.
807	12,528483	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material dblue not found in model database.
808	12,528508	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material dgreen not found in model database.
809	12,528533	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material gray not found in model database.
810	12,528561	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
811	12,537805	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
812	12,537902	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
813	12,537966	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material blue not found in model database.
814	12,538025	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material yellow not found in model database.
815	12,538080	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material pink not found in model database.
816	12,538132	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material cyan not found in model database.
817	12,538183	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
818	12,538232	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material white not found in model database.
819	12,538283	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material dblue not found in model database.
820	12,538333	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material dgreen not found in model database.
821	12,538382	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material gray not found in model database.
822	12,538431	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
823	12,538478	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
824	12,538528	ERROR	  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.

~~~

DanielePucci, 2017-02-22:
I remember that I've seen this errors on simulations run by either @pi-q or @gabrielenava. Please, open issues asap when you notice something fishy.

traversaro, 2017-02-28:
I think it is a leftover of https://github.com/robotology-playground/icub-model-generator/issues/36 . 
For the time being, as a workaround I think we should suppressed this errors in iDynTree as already done for similar errors in https://github.com/robotology/idyntree/commit/5e760abb90642899d81db803b3e653343c74bb4f .

traversaro, 2017-02-28:
In particular we should suppress this line https://github.com/robotology/idyntree/blob/5e760abb90642899d81db803b3e653343c74bb4f/src/model_io/urdf/src/URDFSolidShapesImport.cpp#L105 . 
@diegoferigo If you want you can take care of this, to start familiarizing with the iDynTree codebase.

diegoferigo, 2017-07-04:
Instead of totally suppressing the message I'd rather to switch to a `reportWarning`. Comments?

traversaro, 2017-08-10:
Duplicate of https://github.com/robotology-playground/icub-model-generator/issues/62 .

closed 2017-08-10 by traversaro