robotology-legacy/codyco-modules #223: Deprecate wholeBodyDynamicsTree

traversaro, 2017-02-22:
Before CoDyCo Y4, I would like to make sure that all codyco users stopped using `wholeBodyDynamicsTree` and started using the new wholeBodyDynamicsDevice.

traversaro, 2017-02-24:
First step: 
 * https://github.com/robotology/codyco-modules/pull/224

Missing steps: 
 * Provide files for bigman and its variants.
 * Add files for `floatingBaseEstimator` .

traversaro, 2017-03-29:
Blocked by https://github.com/robotology/codyco-modules/issues/227 .

traversaro, 2017-04-21:
Relevant docs:  https://github.com/robotology/codyco-modules/blob/master/doc/force_control_on_icub.md .

traversaro, 2017-05-19:
Apparently in wholeBodyDynamicsTree  there is some legacy port that are giving in output the com position (that are used by some users such as @serena-ivaldi ). This kind of functionality have been migrate to the simpleLeggedOdometry device, but we should make sure to support those ports there before deprecating wholeBodyDynamicsTree .

traversaro, 2017-08-10:
Final missing feature w.r.t. to the old wholeBodyDynamicsTree migrated in https://github.com/robotology/codyco-modules/pull/239, we are ready to merge.

serena-ivaldi, 2017-12-08:
Hi Silvio, 
it is not 100% clear for us how we should interact with the iDynTree suit (considering all code developed using iDynTree) right now... 
We would like to instantiate a model of iCub with iDynTree, provide joints values, and given these values compute some relevant quantities, such as the COM. 
If now this is done by simpleLeggedOdometry (https://github.com/robotology/codyco-modules/blob/master/src/modules/wholeBodyDynamicsTree/src/simpleLeggedOdometry.cpp) we can't see where... maybe it is not in master?

traversaro, 2017-12-08:
The recommended way of extracting kinematics and dynamics informations related to the robot model is to use the `iDynTree::KinDynComputations` class. For an example of its use (including the computation of the center of mass) is provided in https://github.com/robotology/idyntree/blob/ccbf42fe669cdac19d0277eeeaa4e115e83cdd66/examples/KinDynComputationsWithEigen/main.cpp .

serena-ivaldi, 2017-12-08:
thanks - we were still looking into the libraries of codyco-modules, we didn;t know it was now a separated project in robotology...

traversaro, 2019-03-15:
Hi @serena-ivaldi , are you still using the `wholeBodyDynamicsTree` module for some experiments?

traversaro, 2020-02-01:
`wholeBodyDynamicsTree` has not been compiled for a long time.

closed 2020-02-01 by traversaro