robotology-legacy/codyco-modules #225: torque timeout while running wholeBodyDynamicsDevice throught yarprobotinterface

traversaro, 2017-02-28:
Today on `iCubGenova02` we experienced the following failure:

~~~
12976	4937,704660	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
12977	4938,655083	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
12978	4938,715888	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
12979	4939,676715	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
12980	4939,737185	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
12981	4940,433463	ERROR	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 5096s 254m 359u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
12982	4940,433529	ERROR	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 5096s 254m 503u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
12983	4940,433550	ERROR	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 5096s 254m 637u: (code 0x02000007, par16 0x0002 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
12984	4940,433567	ERROR	 from BOARD 10.0.1.9 (EB9), src LOCAL, adr 0, time 5096s 271m 344u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
12985	4940,433583	ERROR	 from BOARD 10.0.1.9 (EB9), src LOCAL, adr 0, time 5096s 271m 493u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
12986	4940,433605	ERROR	 from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 5096s 232m 116u: (code 0x02000007, par16 0x0000 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
12987	4940,433623	ERROR	 from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 5096s 232m 265u: (code 0x02000007, par16 0x0001 par64 0x0000000000000000) -> MC: torque sensor timeout. The joint is in a compliant interaction mode or torque control mode, and data from torque sensor have been unavailable for more than 100 ms. par16 = ID of joint. + .
12988	4940,433639	WARNING	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 5096s 254m 949u: (code 0x0000000d, par16 0x018c par64 0x0004001f00350233) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
12989	4940,686341	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
12990	4940,747019	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
12991	4941,708888	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
12992	4941,769570	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet

~~~

traversaro, 2017-04-20:
We had a discussion with @valegagge back in march. She told me that they in the past they also experienced problems in the communication between the PC104 and the EMS, so the problem could also be related to that.

DanielePucci, 2017-04-20:
This problem is happening again: @pi-q @gabrielenava @S-Dafarra are going to provide support to @valegagge to care of this problem (e.g. post the logs of the recent occurrences)

traversaro, 2017-04-20:
As this is probably a system problem, related to the interaction of the `wholeBodyDynamicsDevice` with the rest of the iCub ETH interface, I think it make more sense to move this issue to `icub-support`.

traversaro, 2017-04-20:
Issue moved to [robotology/icub-support #449](https://github.com/robotology/icub-support/issues/449) via [**ZenHub**](https://www.zenhub.com/)

closed 2017-04-20 by traversaro