robotology-legacy/codyco-modules #233: wholebodydynamics.xml not compliant with the new device naming system

S-Dafarra, 2017-05-30:
The same as https://github.com/robotology/human-dynamics-estimation/issues/67 .
See https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-eth-six-fts.xml#L44 vs https://github.com/robotology/robots-configuration/blob/master/iCubGenova04/wrappers/motorControl/right_leg-mc_wrapper.xml#L21

S-Dafarra, 2017-05-30:
A possible patch is the following:

```
        <action phase="startup" level="15" type="attach">
            <paramlist name="networks">
                <!-- motorcontrol and virtual torque sensors -->
                <elem name="left_lower_leg">left_leg-eb7-j4_5-mc</elem>
                <elem name="right_lower_leg">right_leg-eb9-j4_5-mc</elem>
                <elem name="left_upper_leg">left_leg-eb6-j0_3-mc</elem>
                <elem name="right_upper_leg">right_leg-eb8-j0_3-mc</elem>
                <elem name="torso">torso-eb5-j0_2-mc</elem>
                <elem name="right_lower_arm">right_arm-eb27-j4_7-mc</elem>
                <elem name="left_lower_arm">left_arm-eb24-j4_7-mc</elem>
                <elem name="right_upper_arm">right_arm-eb3-j0_3-mc</elem>
                <elem name="left_upper_arm">left_arm-eb1-j0_3-mc</elem>
                <elem name="head-j0">head-eb20-j0_1-mc</elem>
                <elem name="head-j2">head-eb21-j2_5-mc</elem>
                <!-- imu -->
                <elem name="imu">inertial</elem>
                <!-- ft -->
                <elem name="l_arm_ft_sensor">left_arm-eb1-j0_3-strain</elem>
                <elem name="r_arm_ft_sensor">right_arm-eb3-j0_3-strain</elem>
                <elem name="l_leg_ft_sensor">left_leg-eb6-j0_3-strain</elem>
                <elem name="r_leg_ft_sensor">right_leg-eb8-j0_3-strain</elem>
                <elem name="l_foot_ft_sensor">left_leg-eb7-j4_5-strain</elem>
                <elem name="r_foot_ft_sensor">right_leg-eb9-j4_5-strain</elem>
            </paramlist>
        </action>
```
@traversaro can you please double check?

traversaro, 2017-05-30:
It seems correct. 
In the medium term we should find a way to encapsulate such complexity from specific device configuration files. 
Possible solutions include: 
 * supporting "group" of devices declared somewhere in the yarprobotinterface configuration files   
  * attaching only "remapped" devices that contain all the resources of a given robot. We can do that for the controlboard devices, but at the moment a "remapper" device is only planned for IMUs and totally missing for FTs .