robotology-legacy/codyco-modules #236: Add a new parameter to wholeBodyDynamics

traversaro, 2017-07-28:
Originally posted by @fjandrad  : 
> How should I add a new configuration parameter to wholeBodyDynamics? I ask because I see you have a function for almost every group or parameter that is not in a group and also one for each group? Should I also have a function for loading the flag useSkinContacts?
> Also is there a template for that file wholebodydynamics.xml? Where is it stored?
> I see there is a autogenerated file called wholeBodyDynamicsSettings should the variable added somehow there? if thats the case how and where can I do it?

traversaro, 2017-07-28:
In order: 
 * I misunderstood a bit the original request. For reading just the skin contact positions, I think it is ok to use them if they are available, as it was the behaviour of `wholeBodyDynamics` and `wholeBodyDynamicsTree` modules, so we do not need an additional option. 
 * For reading the contact F/T estimated from the Skin, I think it is necessary to have an option, because while it will be useful for our experiment to read the contact F/T from the skin for estimating the joint torque, I think the contact F/T  estimate obtained using the internal F/T will remain the default choice for a long time. In this case to add the parameter you need to add it to the documentation in https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.h#L120 and add the appropriate parsing code in https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L559 (the parsing code is divided in several functions just for convenience, a single parameter can easily be added to the generic `loadSettingsFromConfig` method. 
* Once you add a new parameter, you can set its default value to `false`, and in that case you don't need to modify the existing configuration files that are automatically copied in robot-specific directories. 
If you indeed need to add a configuration parameter, you can add it to all the `*.xml` files in https://github.com/robotology/codyco-modules/tree/master/src/devices/wholeBodyDynamics/app that start with `wholebodydynamics-`.  To read a bit more about this, check https://github.com/robotology/codyco-modules/blob/master/doc/force_control_on_icub.md .
* Regarding the autogenerated wholeBodyDynamicsSettings file, it was meant to give a easy way to change the configuration parameters of the model (and read the current parameters) of the module on the fly over a a YARP RPC port, see https://github.com/robotology/yarp/issues/184 . At the moment the RPC functionality is not really user friendly, but if you think that this parameter can be changed at runtime (i.e. you can handle switching it from on to off between two different estimation cycles) it makes sense to add this parameter to wholeBodyDynamicsSettings. You can do that by modifying the original thirft file at https://github.com/robotology/codyco-modules/blob/master/src/idl/wholeBodyDynamicsSettings/wholeBodyDynamicsSettings.thrift . To read more about thrift, check http://www.yarp.it/idl_thrift.html , and http://www.yarp.it/thrift_editor.html regarding how to edit structure remotely.

cc @diegoferigo that could be interested as well by this software technicalities.

fjandrad, 2017-08-09:
Regarding "For reading the contact F/T estimated from the Skin, I think it is necessary to have an option, because while it will be useful for our experiment to read the contact F/T from the skin for estimating the joint torque, I think the contact F/T estimate obtained using the internal F/T will remain the default choice for a long time."

It will be dealt with the integer variable trustSkinThreshold. So that even if a skinContact exist the force and moment information will not be used for the estimation only the location depending on how reliable is the calibration of the skin part.