robotology-legacy/codyco-modules #249: torqueBalancing ini files not being installed

fjandrad, 2017-10-16:
The CMakeList in 

> codyco-modules/src/modules/

contains the following condition
```
if(CODYCO_USES_KDL)
add_subdirectory(torqueBalancing)
```

Since KDL is being deprecated by default it wont compile the torqueBalancing folder. 
This is an issue since the ini files from torqueBalancing are commonly used for the starting positions.
As a result the ini files are not installed by default.

fjandrad, 2017-10-16:
cc @diegoferigo  @traversaro

traversaro, 2017-10-16:
It was my fault, sorry guys. It can be easily fixed by moving the `if(CODYCO_USES_KDL)` inside the `torqueBalancing` directory, guarding everything except the `add_subdirectory(app)` call.

traversaro, 2017-11-30:
Apparently I forgot to fix this issue. Solution provided in https://github.com/robotology/codyco-modules/commit/5442a76c1a4c2bdb6bf4382432171423d6c81075 . 
cc @diegoferigo @gabrielenava

closed 2017-11-30 by traversaro