robotology-legacy/codyco-modules #256: wholebodydynamics.xml - Cannot terminate yarprobotinterface when using iCubGazeboV2_5

aikolina, 2018-01-23:
After running the following command:
```bash
YARP_ROBOT_NAME=iCubGazeboV2_5 yarprobotinterface --config launch-wholebodydynamics.xml
```
I do all my test and, when I'm done, I try to terminate the `yarprobotinterface` with `CTRL+C`. It seems that everything is shut down properly since I get the messages:
```bash
...
[INFO]shutdown phase finished.
[INFO]RFModule finished.
```
But actually the program does not terminate completely and  I have to `CTRL+C` multiple times until the terminal is released.

**NB** Before typing a `CTRL+C` for the second time, I let `yarprobotinterface` stay for a while (time to take a coffee and come back) hoping that it was just slow in closing everything, but it was not the case.

gabrielenava, 2018-01-23:
I never noticed it, probably because I usually press `CTRL+C` several times. But I confirm that I observed the same behaviour on my pc now.

diegoferigo, 2018-01-25:
I remember it happened also here sometimes. But after a while my compulsive Ctrl+C made me forget about this :sweat_smile:

dosh1234, 2018-07-30:
@aikolina I have a problem. Can you help me ! When I run 'YARP_ROBOT_NAME=iCubGazeboV2_5 yarprobotinterface --config launch-wholebodydynamics.xml' . I got  "did not find launch-wholebodydynamics.xml" , how  can I get the launch-wholebodydynamics.xml. I have install YARP and iCub-main and so on! Thank you!

aikolina, 2018-07-30:
Hi @dosh123, I think your problem is different from the one I describe here. Can you please open a new issue and describe your problem in details? Thanks!