robotology-legacy/codyco-modules #257: Compilation problems since with newest version of ``yarp`` ``devel``

S-Dafarra, 2018-01-26:
I get the following error:
```
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp: In member function ‘virtual bool yarp::dev::VirtualAnalogRemapper::attachAll(const yarp::dev::PolyDriverList&)’:
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:87:58: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘getChannels’
         int nrOfVirtualAxes = virtualAnalogList[devIdx]->getChannels();
                                                          ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:134:89: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘getChannels’
         remappedSubdevices[subdev].measureBuffer.resize(remappedSubdevices[subdev].dev->getChannels(),0.0);
                                                                                         ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:135:91: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘getChannels’
         remappedSubdevices[subdev].local2globalIdx.resize(remappedSubdevices[subdev].dev->getChannels(),-1);
                                                                                           ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:156:79: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘getChannels’
         if( axesOfSubDeviceThatAreRemapped == remappedSubdevices[subdev].dev->getChannels() )
                                                                               ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp: In member function ‘virtual bool yarp::dev::VirtualAnalogRemapper::updateMeasure(yarp::sig::Vector&)’:
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:224:28: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘updateMeasure’
             bool ok = dev->updateMeasure(this->remappedSubdevices[subdevIdx].measureBuffer);
                            ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:239:32: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘updateMeasure’
                 bool ok = dev->updateMeasure(localAxis,measure[jnt]);
                                ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp: In member function ‘virtual bool yarp::dev::VirtualAnalogRemapper::updateMeasure(int, double&)’:
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:262:20: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘updateMeasure’
         ret = dev->updateMeasure(localAxis,measure);
                    ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp: In member function ‘virtual int yarp::dev::VirtualAnalogRemapper::getState(int)’:
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/virtualAnalogRemapper/VirtualAnalogRemapper.cpp:290:20: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘getState’
         ret = dev->getState(localAxis);
                    ^
src/devices/virtualAnalogRemapper/CMakeFiles/virtualAnalogRemapper.dir/build.make:86: recipe for target 'src/devices/virtualAnalogRemapper/CMakeFiles/virtualAnalogRemapper.dir/VirtualAnalogRemapper.cpp.o' failed
make[5]: *** [src/devices/virtualAnalogRemapper/CMakeFiles/virtualAnalogRemapper.dir/VirtualAnalogRemapper.cpp.o] Error 1
CMakeFiles/Makefile2:705: recipe for target 'src/devices/virtualAnalogRemapper/CMakeFiles/virtualAnalogRemapper.dir/all' failed
make[4]: *** [src/devices/virtualAnalogRemapper/CMakeFiles/virtualAnalogRemapper.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
[ 57%] Built target torqueBalancing
CMakeFiles/Makefile2:650: recipe for target 'src/devices/virtualAnalogClient/CMakeFiles/virtualAnalogClient.dir/all' failed
make[4]: *** [src/devices/virtualAnalogClient/CMakeFiles/virtualAnalogClient.dir/all] Error 2
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp: In member function ‘bool yarp::dev::WholeBodyDynamicsDevice::openRemapperVirtualSensors(yarp::os::Searchable&)’:
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp:245:66: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘getChannels’
     int axes = remappedVirtualAnalogSensorsInterfaces.ivirtsens->getChannels();
                                                                  ^
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp: In member function ‘void yarp::dev::WholeBodyDynamicsDevice::publishTorques()’:
/home/sdafarra/Software/codyco-superbuild/main/codyco-modules/src/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp:1662:61: error: ‘class yarp::dev::IVirtualAnalogSensor’ has no member named ‘updateMeasure’
     this->remappedVirtualAnalogSensorsInterfaces.ivirtsens->updateMeasure(this->estimatedJointTorquesYARP);
                                                             ^
```

It is probably due to https://github.com/robotology/yarp/commit/2628fde78f527fd72dfdce37dba6e0851c7d957c#diff-aa2670150d8480ce5105d7333a2ebe17

S-Dafarra, 2018-01-26:
cc @traversaro @gabrielenava

traversaro, 2018-01-26:
Sorry, we introduced the `devel` branch also on codyco-modules, I need to do a proper announcement about that.

closed 2018-01-26 by S-Dafarra