robotology-legacy/codyco-modules #258: [torqueBalancing]-Drift of the robot movements on its left

lpenco, 2018-01-30:
I'm using the torqueBalancing Controller. I'm sending some jointangles references to the robot together with the corresponding com position. I extrapolate the com position through DynTree w.r.t the root link.
When I simulate the robot tends to bend the torso to the left and to open the left arm as if the com reference was more on the left.  In what coordinate frame the torqueBalacing takes the com references (root_link and which orientation?)??? Or you think it might be a problem related to the FT sensors?

lpenco, 2018-02-02:
I was using the C++ old version of the torqueBalancing controller that is affected by unstable zero-dynamics

closed 2018-02-02 by lpenco

traversaro, 2018-02-02:
For more info about the zero dynamics problems of the old controlller, see https://arxiv.org/pdf/1603.04178.pdf .