robotology-legacy/codyco-modules #264: Specify used F/T sensors in wholebodydynamics with an explicit parameter

traversaro, 2018-04-18:
TODO 

cc @gabrielenava @ahmadgazar remember me to update this issue

gabrielenava, 2018-04-19:
ping :-)

ahmadgazar, 2018-04-19:
I am sorry I don't get what do you mean by the explicit parameter of the F/T sensors? @traversaro

fjandrad, 2018-08-28:
I think this is meant to allow to enable/disable and FT sensor from the perspective of the estimation without the need to modify the urdf model.

traversaro, 2018-08-28:
Exactly, something like having a  remapper for `SensorsList` (I think there was a similar requirement coming from Andy guys) instead of using all the FT sensors specified in the URDF.

fjandrad, 2018-09-05:
## Procedure to remove sensor at the foot
Remove the **sensor** tag in the the urdf depending on the sensor to be eliminated `l_foot_ft_sensor` or `r_foot_ft_sensor`.
 In this example it will be the left foot sensor.
Since at the foot, there is no other joint after the sensor, wholebodydynamics will collapse the following links to the parent of the ft sensor, the `l_ankle_2` link.
Due to this all references to` l_foot` or `l_sole` as a link should be changed to `l_ankle_2` in the `wholebodydynamics.xml` file this includes the skindynlib section but more importantly the following lines:

```
 <param name="/wholeBodyDynamics/left_foot/cartesianEndEffectorWrench:o">(l_foot,l_sole,l_sole)</param>

 <param name="l_foot">(l_foot_dh_frame,5,5)</param>

```
The attach part of the ft sensor should also be commented
`
 <elem name="l_foot_ft_sensor">left_leg-eb7-j4_5-strain</elem>`

traversaro, 2018-09-12:
It is is difficult for me to address this in the short term, moving from Assignee: Traversaro to Status: Help Wanted.

fjandrad, 2018-09-16:
Considering I'll dive in into wholebodydynamics device to include the temperature section I might as well take a look at this since is useful for testing ft sensor configurations quickly.
 @traversaro we could use the meeting on Monday so you can tell me how you were planning on doing this, to get a head start

traversaro, 2018-09-16:
Yes!

fjandrad, 2019-03-28:
cc @gabrielenava