robotology-legacy/codyco-modules #270: yarprobotointerface went in segfault when the walking controller is closed

GiulioRomualdi, 2018-10-04:
When the [walking controller](https://github.com/robotology/walking-controllers/tree/demos) is closed the the yarprobotinterface went in SEGFAULT. This happens on the iCubGenova02 when the contact wrench ports are disconnected.   

```
13870	1764,926507		yarp: Sending output from /wholeBodyDynamics/left_foot/cartesianEndEffectorWrench:o to /walking-coordinator/leftFootWrench:i using tcp
13871	1764,936514		yarp: Sending output from /wholeBodyDynamics/right_foot/cartesianEndEffectorWrench:o to /walking-coordinator/rightFootWrench:i using tcp
13877	1765,057765		yarp: Removing output from /wholeBodyDynamics/right_foot/cartesianEndEffectorWrench:o to /walking-coordinator/rightFootWrench:i
13878	1765,057799		*** Error in `yarprobotinterface': free(): invalid size: 0x00007fd1b8001f80 ***
13879	1765,057811		======= Backtrace: =========
13880	1765,057819		/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fd1c651f7e5]
13881	1765,057828		/lib/x86_64-linux-gnu/libc.so.6(+0x7fe0a)[0x7fd1c6527e0a]
13882	1765,057837		/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fd1c652b98c]
13883	1765,057845		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(_ZN4yarp2os13NestedContactD2Ev+0x68)[0x7fd1c6eb5f08]
13884	1765,057853		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(_ZN4yarp2os7ContactD2Ev+0x21)[0x7fd1c6e760b1]
13885	1765,057862		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(_ZN4yarp2os5RouteD1Ev+0x33)[0x7fd1c6f1c603]
13886	1765,057871		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(+0xda4e5)[0x7fd1c6ee34e5]
13887	1765,057880		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(_ZN4yarp2os4impl8PortCore10cleanUnitsEb+0x132)[0x7fd1c6ed22e2]
13888	1765,057889		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(_ZN4yarp2os4impl8PortCore9reapUnitsEv+0x127)[0x7fd1c6ed2727]
13889	1765,057897		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(_ZN4yarp2os4impl8PortCore3runEv+0x118)[0x7fd1c6edec38]
13890	1765,057905		/usr/local/src/robot/yarp/build/lib/libYARP_OS.so.3(+0x12538d)[0x7fd1c6f2e38d]
13891	1765,057913		/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80)[0x7fd1c6b3fc80]
13892	1765,057921		/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7fd1c55306ba]
13893	1765,057931		/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7fd1c65ae82d]
```
[yarprobotinteface_segfault.txt](https://github.com/robotology/codyco-modules/files/2446192/yarprobotinteface_segfault.txt)

traversaro, 2019-03-15:
Is this a reproducible problem? It happens all the time? You experienced it also more recently?

traversaro, 2020-02-01:
> Is this a reproducible problem? It happens all the time? You experienced it also more recently?

Friendly ping @GiulioRomualdi, if this never happened again I think we can close the issue.

traversaro, 2020-04-05:
Given the lack of response, I would go ahead and assume that the problem is not appearing anymore. If the problem arise again, please open a new issue in https://github.com/robotology/whole-body-estimators, thanks!

closed 2020-04-05 by traversaro